EFeru/hoverboard-firmware-hack-FOC #350: Source/supplier for compatible mainboards?

derFrickler, 2022-10-25:
### Variant

HOVERCAR

### Description

Hi there, I have already build some stuff with this great firmware in the past, but now i am struggling to find a compatible mainboard somewhere. 

There are nearly none of them sold on ebay (Germany) anymore and on other sources like Aliexpress i only see the black versions which are not compatible i guess? https://www.aliexpress.com/item/1005003267774605.html

It would be great if someone has a source where to get boards that are compatible with this firmware - might help other people as well. 

thanks a lot for the great work!

Candas1, 2022-11-01:
Hi,

You can try buying cheap used hoverboards but there is a chance you will end up with split boards.
There are [Telegram group](https://t.me/joinchat/BHWO_RKu2LT5ZxEkvUB8uw) members that might sell some of those boards in Germany.

The cheap boards on aliexpress/ebay are the [Artery boards](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Firmware-Compatibility#artery-boards).
You can also show your interest [here](https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/329)

derFrickler, 2022-11-11:
Thanks for pointing me to the telegram group. They tried to help me repairing my board, but without success so far. 

But i just got 2 Boards from ebay (DE) and they work fine. There are some more left if you are interested:  ,
https://www.ebay.de/itm/155234924916

closed 2022-11-17 by derFrickler