EFeru/hoverboard-firmware-hack-FOC #355: USART lag issue

ChristopheLaurent, 2022-11-05:
### Variant

USART

### Control type

FOC

### Control mode

Speed

### Description

Hello,

I manage to drive my board motors from an ESP32 using USART, with a code similar to [hoverserial.ino](https://github.com/EFeru/hoverboard-firmware-hack-FOC/tree/main/Arduino/hoverserial) but using HardwareSerial instead of SoftwareSerial, and I'm using tank steering. Another difference is in my main loop that has to do other things than sending commands to the hoverboard periodically, so basically my main loop looks like:

void loop()
{
  // do something
  CheckWheels(); // performs reading of feedback via USART3, and log it with current time
  MoveWheels(_speedLeft, _speedRight); // send speed command via USART3, and log it with current time
  delay(10); // sleep 10ms to give time to the interruptions...
}

I disabled the filter in the config file so I expect to see my command (speeds for left and right wheels) in the [feedback fields Cmd1 & Cmd2](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-USART#speech_balloon-feedback). I do, but exactly 15 samples later. Since I have a delay of 10 ms in my loop, my command is took in account by the hoverboard firmware 150 ms later.
I did log the commands and the feedbacks in a csv file and plot that in excel:
[image]
The horizontal axis is time in ms, and the vertical axis is speed in RPM. The blue dots are the right wheel commands, the orange dots are the right wheel commands from the feedback, and the gray dots are the right wheel measured speeds from the feedback .
In the first part of the chart, before 4s, the setup() sends a specific speed profile with a delay of 20ms between the commands. I can find each command in the feedback, but 300ms later. In the second part of the chart, I drive the board manually with a joystick (from another ESP32 via ESPNOW) with a delay of 10ms between commands. I can find each command in the feedback, but 150ms later. This is clearly visible around 7s:
[image]
Once the command is took in account by the hoverboard firmware (orange dot), the speed control loop works pretty well and the measured speed (gray dots) follows closely the command. Unfortunately, I have a 150ms lag between the blue dots and the orange dots.

Now the questions:

- Is this lag a bug, a feature, or a side effect of the speed control loop ?
- What is the optimal rate for sending commands via USART ?

Cheers !

Candas1, 2022-11-06:
Hi,

You checked if it's due to RATE parameter?

ChristopheLaurent, 2022-11-07:
Hi Candas,
Yes I did check it. I did not change the Rate factor and you can see in above charts that the speed increase is below 30 RPM threshold. Looking at the hoverboard firmware source code, Rate parameter is applied on the feedback Cmd1 & Cmd2, so if it was applied I would see it on this feedback.
I eventually managed to mitigate the issue by reducing the sleep time in my main loop, but it is still a concern.
Thank you for your support & best Regards,
Christophe

Candas1, 2022-11-07:
So if FILTER and RATE are ok, I don't think there is a bug.

You cannot expect that it happens instantly.
I think the arduino sketch sends and receives every 100ms.
The firmware main loop runs every 5ms but it sends feeback only every 10ms.

ChristopheLaurent, 2022-11-11:
Thank you for your feedback, but my ESP32 did not send and receive every 100ms but 10ms.
Here are the details of the samples sent and received corresponding to the sequence of my second chart above:
[image]
Let me explain:

- At 7174ms from the beginning of the Arduino sketch, I send 44 RPM command (Note that I sent before 37 RPM so there is no RATE issue). 
- At 7185ms, I read 11 RPM in the Feedback field Cmd1 and the measured speed is 0 RPM.
- At 7185ms, I send 49 RPM command.
- At 7196ms, I read 16 RPM in the Feedback field Cmd1 and the measured speed is 5 RPM. This means that 22ms after sending 44 RPM command, I still do not see my command in the feedback field. Is it lost ? No, it is late...
- At 7328ms, I eventually read 44 RPM in the Feedback field Cmd1, 154ms after sending the command.

You can clearly see this shift of 15 samples between the command and the feedback in this listing. If I increase the time between each command to 50ms, the lag is no longer 150ms but 15x50ms=750ms. If I increase the time between each command to 100ms, the board start beeping due to the timeout on USART (according to your excellent documentation).
Do you have another explanation for this lag between the Command and the Feedback ?

If you don't, then don't worry. I managed to optimize my Arduino sketch so I can send the commands every 1 ms. The observed lag decreased to 15ms which is acceptable for my application. I just hope it won't become a bigger issue later.

Best Regards.

Candas1, 2022-11-14:
The few times I used USART Variant, timing wasn't critical for my use case.
Are samples are shifted ? or skipped ? Based on your graph it looks like samples are shifted by 15 samples, you don't seem to miss messaged ?
For this to happen, samples should be somewhere a buffer, but I don't know if this happend when:
- sending the command
- receiving the command
- sending the feedback
- receiving the feedback

I don't have time now but I would need to try to reproduce this, to understand if it's due to your setup or if it's a bug.

ChristopheLaurent, 2022-11-15:
Samples are shifted, nothing is skipped or lost. No messages missed. You're right, samples should be somewhere in a buffer, and I guess it happens between receiving the command and sending the feedback. May the speed control algorithm (PID ?) require past samples ?
I don't have any regular Arduino to run hoverserial.ino, but it should be possible to reproduce it with a couple of changes in this sketch. I don't have time now but I will try next month and let you know. Thanks for your support.

Candas1, 2022-11-15:
You are comparing the command you send and the command from the feedback right ?
Pid is part of motor control and has only impact on the measure speed.

[Here](https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/24b3d2515f70c0709b658364930930a935e6552c/Src/main.c#L510) we directly send the feedback after mapping/filtering/rate/mixing.