EFeru/hoverboard-firmware-hack-FOC #356: Will not start at cold

jusubbi, 2022-11-18:
### Variant

HOVERCAR

### Control type

FOC

### Control mode

Torque

### What is the bug and how to reproduce it ?

I noticed now in the winter that hoverboard won't start in the cold. When I start from a warm garage, no problems, but when I go to the store and try to start, it won't start all the way. There are a few beeps, but it gets stuck at that point. So there will be no final beeps. I took the device into the store for a while and it started again. The fault is not in the battery. There is a lot of voltage. How does the software examine the temperatures during startup and could it be disabled completely? I mean 0 and below  (minus) temperatures celsius.

Candas1, 2022-11-18:
Hi,

Firmware doesn't use the temperature. If no final beep, probably it's the calibration. Check the wiki.

jusubbi, 2022-11-18:
But it's related to temperature somehow. There is lot of temperature ADC values calibration etc. at startup.

jusubbi, 2022-11-18:
Maybe I try auto calibration next time when this problem occurs. I bet it won't work neither.

Candas1, 2022-11-18:
Yes it could be temperature is impacting adc reading. I am wondering if this could be compensated by software.
But for now you need to calibrate when this happens

jusubbi, 2022-11-22:
You were right. Problem caused inputs adc. When it was cold, it stops here:

Using the configuration from config.h
Limits Input1: TYP:1 MIN:940 MID:0 MAX:3100
Limits Input2: TYP:1 MIN:1010 MID:0 MAX:3120
Drive mode 0 selected: max_speed:1000 acc_rate:450

After it warm up, then it starts to read values:

-- Motors enabled --
in1:945 in2:1004 cmdL:0 cmdR:0 BatADC:1873 BatV:4902 TempADC:1768 Temp:138
in1:948 in2:1005 cmdL:0 cmdR:0 BatADC:1867 BatV:4899 TempADC:1754 Temp:165
in1:949 in2:1007 cmdL:0 cmdR:0 BatADC:1875 BatV:4899 TempADC:1744 Temp:184
etc.

I made auto calibration at warm place and it made these modifications:

Input calibration started...
Input1 is a normal pot AND protected..OK
Input2 is a normal pot AND protected..OK
Limits Input1: TYP:1 MIN:1043 MID:944 MAX:2980
Limits Input2: TYP:1 MIN:1105 MID:1006 MAX:2995

After that there is no problem in cold enviroment. Problem solved, thank you!

closed 2022-11-22 by jusubbi