EFeru/hoverboard-firmware-hack-FOC #357: GD32F303RCT6 firmware hack hoverboard.ino don't work

AnOrdinaryUsser, 2022-11-21:
### Variant

USART

### Control type

Commutation

### Description

Good morning to you all,

I am trying to use, a GD32F303RCT6 processor with a tao tao board. This:

[image]
[image]

I unlock the motherboard and upload the firmware but But I am not able to use the hoverboard.ino arduino skecth to test the wheels. My problem is that I don't know if I'm misusing the RX and TX pins on the board or if I need to modify something to make it work properly.

Candas1, 2022-11-21:
Hi,

Have you checked the troubleshooting hints in the wiki ?
I think someone had reported about a board with a F303 chip in the telegram channel, it seemed to work

Candas1, 2022-11-21:
https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/191

AnOrdinaryUsser, 2022-11-21:
> Hi,
> 
> Have you checked the troubleshooting hints in the wiki ? I think someone had reported about a board with a F303 chip in the telegram channel, it seemed to work [edit] Actually there is no proof this board would work, pinout probably has to be checked, and also if there major differences between F103 and F303

What is the telegram channel to see those messages?

Candas1, 2022-11-21:
Check the links at the end of the main page of the repository.

Candas1, 2023-03-01:
So basically F303 is not supported and it is a lot of work to rewrite the firmware for this chip family

closed 2023-03-01 by Candas1