EFeru/hoverboard-firmware-hack-FOC #359: hovercar mainboard dead again?

hbnewbie, 2022-12-16:
### Variant

HOVERCAR

### Control type

FOC

### Control mode

Torque

### Description

was hoping to get some help here as i am trying to build a gokart as a christmas gift. am pretty much following the basic instructions for the hovercar variant.  am using stm mainboard and sideboard. have found after flashing the sideboard and mainboard that the battery indicator led keeps flashing orange, has faint constant beeping and hoverboard wont turn off when shorting the power button wires. in addition can see spark inside connector when connecting battery pack.  reading troubleshooting section it looks like its most likely blown mosfet/s.  this is my second mainboard i am trying from a working hoverboard.  on the first one i had tried with all 4 pins from stlink connected to mainboard, without battery pack connected.  this time i followed the instructions with 3.3 pin not connected to mainboard and with battery pack connected with the same result. on this second board i checked resistance on mosfets from drain to source pins and found some of them showing around 9k ohms.  this was before i flashed the firmware. after flashing i checked them again and found what seems to be same results. on every other mosfet am seeing about 9k ohms from drain to source.  have tried diode mode as described from google tests with positive lead to gate and negative to source but found no change.  i think my meter may not provide enough voltage to do that test. im wondering if anyone else has encountered this issue or if they know why this might be happening?

Candas1, 2022-12-16:
Hi,

Sorry I don't understand your issue.
Is it just sparks when plugging in the battery or is it burning your connector ?

hbnewbie, 2022-12-16:
it just sparks once.   possible this is normal but i didnt notice it before flashing.  i should mention i did the flashing in platformio and it always says success.  symptoms are it will not power off until unplugging battery, constant faint beeping, and battery led flashing orange. when i hold power button wires together to short them for over 5 seconds, it also doesnt beep to indicate it is entering auto calibration

hbnewbie, 2022-12-16:
also the normal power up sound works when i short the power switch wires to turn it on

Candas1, 2022-12-16:
It will always spark if you connect the battery and the capacitors are empty. Specially if your board is not turning off, it's emptying the capacitors. What's the voltage of your battery?

hbnewbie, 2022-12-17:
41.7 volts.

hbnewbie, 2022-12-17:
not sure what im doing wrong.  the only things ive changed in config.h are:
#define CTRL_MOD_REQ    TRQ_MODE
#define FIELD_WEAK_ENA  1
// #define SIDEBOARD_SERIAL_USART3 1 <-i think the // means comment out?  i marked this out because i dont want to use the photosensor buttons for anything. 
i have one more non working hoverboard to use that has a possible dead battery.  i can pop a battery in it to see if its working but even if it does, i fear the same thing will happen when i flash it.

Candas1, 2022-12-17:
But is the board also turning on directly when you plug the battery?

Candas1, 2022-12-17:
You said you cannot enter Auto-calibration, it could be the pin for the button is different, or is not working
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Firmware-Compatibility#pin-mapping

hbnewbie, 2022-12-18:
the board turns on normally and has normal startup sound. after plugging in battery it starts by shorting power button wires together. after that it does not turn off if i short the wires together again.  i have found if i connect/disconnect the electric pedal wires i sometimes get erratic behavior and motors start moving slowly or just take off at full speed.  this is the only time i can get them to move.  i have verified 3.3 volts to the hall effect pedals as well as ground, and with pedal not depressed i get .8 volts going to the board and at full throttle i get 2.5 volts from each pedal.  i have also verified the pedals work with a brushless motor controller.  at this point i think something in the board has failed and will likely just use the brushless motor controllers

Candas1, 2022-12-18:
This is why calibration is required.
You cannot do the autocalibration because you cannot get the button to work, but you could still do manual calibration if you have a ftdi.
All this is described in the wiki.

hbnewbie, 2022-12-18:
I was sometimes able to get the button to work to turn it off.  I only have the button cut off so I could keep the wires twisted together while flashing.  Would have soldered it back together after i got it working.  i have tried flashing the alternate firmware version, but then the board doesn't turn on.  So i don't think its a different pin mapping board. I took a picture of the motor phase wires before hover board was taken apart, and it doesn't work with same colors connected, or with various other combinations. I would think even it phases were wrong that the motors would clog or try to move a little.  That was the behavior when connected to my brushles motor controller.  The only time i could get them to move was when disconnecting pedal wires when board was already on.  And that only worked rarely with inconsistent movement. I don't have an ftdi.  I will try entering the pedal voltages into config and reflash again and keep reading more

Candas1, 2022-12-18:
Do you have a good quality picture of your board just in case?

hbnewbie, 2022-12-18:
[image]

hbnewbie, 2022-12-18:
You know I didn't even notice it before, but after you asked to get a picture I noticed the chip has a different number.  I rubbed off the white spot with some alcohol and it's a RET6 and not a RCT6.  Do you think this could be the  cause of the issues? 

[image]

hbnewbie, 2022-12-18:
ok comparing the 2 pinout pictures from some googling does look like the the pinout numbering is different from the pictures:
https://www.kynix.com/components/STM32F103RET6-Embedded---Microcontrollers-CAD-Models-Datasheet-Features.html#stm32f103ret6-pin-configuration
https://www.utmel.com/components/stm32f103rct6-microcontroller-72mhz-64-lqfp-pinout-and-datasheet?id=1242
it looks like the board is probably not a viable option

Candas1, 2022-12-18:
It could be the same chip but with more memory.
Another member in the telegram group had one but he didn't even manage to connect.

Try what is mentioned in the firmware compatibility section of the wiki

hbnewbie, 2022-12-19:
i may mess with that board some more another time, but for now i have moved onto the last hoverboard i have.   i put another battery in and confirmed it starts and runs normally.  i pulled and flashed the mainboard after confirming its a STM32F103RCT6. i changed nothing in the config, just chose hovercar variant and flashed in platformio.  unfortunately its still not working, but symptoms seem to be more consistent this time.  again turns on normally with normal startup sound.  nothing happens when pressing hall effect pedals.  and again board does not turn off until i disconnect the battery.  when the board is on,  if i disconnect the 3.3v power to the pedals, it does the 2 beep indicating ADC protection.  when i plug the power wire back in for the pedals, they both start turning slowly. at this point  if i press the brake pedal it seems to work normally and stops the wheels.  when i double press works normally and goes into reverse and beeps indicating reverse direction.  when i press the gas pedal however it also stops the wheels instead of speeding up.  does not go into reverse when double pressed like the brake pedal does though. i have the pedals wired like the picture for variant hovercar with 100,000 ohm resistors from the signal wires to ground.  and i measured .82 volts when released and 2.5v full throttle for both pedals going to the board after the resistor on left cable blue and green wires.  i have seen mention of pedals doing similar things on this page:https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Firmware-Compatibility#toolbox-troubleshooting . it also mentions finding right hall sensor mapping and current reading which you mentioned here:  https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/230 .  so i guess i will try sinusoidal mode and changing the hall pin numbers first, although this doesnt seem to be a common issue for this board

Candas1, 2022-12-19:
Have you done the auto calibration?

hbnewbie, 2022-12-19:
Have tried auto calibration multiple times.  Doesn't start the process or
give the single beep to indicate it's starting.  I'm guessing it's related
to not turning off when I use the power button again. When I plug the 3.3v
to the pedals back in, the power button works to turn off the board.  Then
when it's off , it again starts normally and does nothing, and again will
not turn off til I unplug the battery. It doesn't beep either. It does not
do the 2 beeps either to indicate ADC timeout until I unplug the 3.3v to
the pedals again.  I should mention the phase wires to the motors are not
matching colors.  I just connected them as they were originally connected
when I took the hoverboard apart. Swapping them around hasnt made a
difference.  The hall wire colors to the motors may also be different on
this other main board.  I haven't had time to check yet will have to get
back on it when I get home from work.

Candas1, 2022-12-19:
even with the different board ?
You must be doing something wrong
the firmware is working for many people

hbnewbie, 2022-12-19:
I agree with you I just don't know what part I could be doing wrong.  I
haven't been able to find any search results with the same symptoms I'm
having but some in the trouble shooting/ diag sections are somewhat
similar, that I have mentioned in previous post. I just haven't had time to
try them yet.  If the hall wires have swapped colors on this board I don't
know if that would shut the motors down entirely and disable power off.
There's nothing else I can think of but I'll try some more when I get home

Candas1, 2022-12-19:
No this is totally unrelated
The button and how the board turns on and off is one thing.
How the inputs are handled is another.
And how the motor is controlled is another.

You need to tackle it separately.

hbnewbie, 2022-12-22:
ok so today i had a little time to play with it and i flashed control type sinusoidal with adc variant.  started up and turned off normally, and was able to enter auto calibration and control wheels.  one pedal accelerated, and the other accelerated wheels in opposite directions.  do you have any idea why it might only be having the issue in hoverboard variant?  i will try to flash again with sinusoidal and see if it makes a difference.  at this point i would be happy with just adc mode, but there doesnt seem to be an option in config for using one pedal as brake or reverse.  i will report back after flashing sinusoidal

hbnewbie, 2022-12-22:
sorry last post i meant hovercar variant not hoverboard. i just flashed again with hovercar variant /sinusoidal /voltage mode and its a no go.  its doing same thing it was doing before.  i noticed input calibration in the stock config file shows hovercar variant pri-input as 1 instead of 3 for auto detect, but changing it made no difference for me.  i should also mention i havent had the sideboard connected for a while, but i dont think it should make a difference to test the motors?

Candas1, 2022-12-22:
Hovercar needs two pedals or handles
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-HOVERCAR#control_knobs-calibration

hbnewbie, 2022-12-22:
I understand I have been using two pedals wired as it is in the hover car
variant page picture.   When I flash ADC variant one of the pedals
accelerates both motors normally.  The other pedal accelerates motors in
opposite directions.

Candas1, 2022-12-22:
Then hovercar should work.
Only calibration is needed

Candas1, 2022-12-22:
Disable multi-mode in config.h

Candas1, 2022-12-22:
[Here](https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/e5f7e9457aa26509f44070b821e35aa3ca705675/Inc/config.h#L528)

hbnewbie, 2022-12-22:
Will try it when I get home.  Thanks

hbnewbie, 2022-12-23:
its working perfectly.  thank you for all the help candas! i dont know how long i wouldve taken to find the cause of this issue, if at all.  i will try that other board again when ihave free time

Candas1, 2022-12-23:
Sorry I didn't pick this up quicker.
It only clicked when someone else mentioned about the same issue.

Hi @EFeru, same issue here.

hbnewbie, 2023-01-01:
[image]
just as an update, i did finish the gokart by christmas and it worked great.  i have tried the STM32F103RET6 board again and it works fine.  i didnt have to do anything different than the instructions.   i have flashed adc variant to it and it works normally except the controls are erratic. i think thats only because of the cheap joysticks i bought to test it.  they dont seem to have a linear voltage output/have drift.  
https://www.amazon.com/Wishiot-Joystick-Controller-Breakout-Arduino/dp/B089VXPHDH/ref=sr_1_18_sspa?crid=2X1VNXGKQHRZZ&keywords=potentiometer+joystick&qid=1672598927&sprefix=potentiometer+joysti%2Caps%2C446&sr=8-18-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUEzQkcxSVZVMEJaMllEJmVuY3J5cHRlZElkPUEwMzIyMjY5M0owQ01OQk5GRjFHJmVuY3J5cHRlZEFkSWQ9QTAwMjA2NzgyTDJMNDc2TjNEQ0pWJndpZGdldE5hbWU9c3BfYnRmJmFjdGlvbj1jbGlja1JlZGlyZWN0JmRvTm90TG9nQ2xpY2s9dHJ1ZQ==
if anyone is thinking about using these i would avoid them

Candas1, 2023-01-03:
Nice build
Good to know STM32F103RET6 also works, I will add it in the wiki
Are you sure the joysticks don't have a linear voltage output ? It can also happen in torque mode

hbnewbie, 2023-01-03:
well i probably shouldnt have described it that way, the voltage does seem linear as it is just potentiometers.  but the joystick is so small, and the range is pretty narrow so its almost like an on off switch.  when measuring the voltage it maxes out well before the stick has reached its end of travel.  i also noticed it doesnt center well.  when i release it the wheels keep spinning slowly until i move it back a little the other direction and then it stops.  but that could probably be fixed by increasing the deadband. they definitely work for testing since theyre so cheap, or for someone better at videos games than me

hbnewbie, 2023-01-05:
https://user-images.githubusercontent.com/120746298/210689959-aeecfeb0-5de1-42c7-a925-ee07b1238e1a.mp4

the RET6 board is working now on a 14s ego lawnmower battery.  i seemed to have the same issue mentioned by gerkon https://candas1.github.io/Hoverboard-Web-Serial-Control/ where the wheels would suddenly jerk to a stop when accelerating, and also switched to sinusoidal control which seems to work fine.  i loved the freewheeling of trq mode so ill give this a shot, and if i really dislike i may switch back to 36v and see how much speed i can get from it.   but anyways i guess this should be closed now? since youve solved all my problems i had issues with.  thanks again for the help candas and eferu for the awesome firmware

hbnewbie, 2023-01-05:
sorry just posted wrong link in last post in case anyone wanted to read about the higher voltage mentioned:
https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/24

closed 2023-01-05 by hbnewbie