EFeru/hoverboard-firmware-hack-FOC #360: Video. 1. Right wheel won't stop. 2. Bugg when turning wheels.

DmitrySurgut, 2022-12-23:
### Variant

HOVERCAR

### Control type

FOC

### Control mode

Torque

### What is the bug and how to reproduce it ?

Video. Right wheel won't stop. Bugg when turning wheels. https://drive.google.com/drive/folders/1LTxlPl3RY7YzgZsmhaCjnE4IPPUD_CdC?usp=share_link

Candas1, 2022-12-23:
Both won't happen with load

Candas1, 2022-12-23:
And if it oscillates you probably have the brake pressed

Candas1, 2022-12-23:
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-HOVERCAR#toolbox-troubleshooting

DmitrySurgut, 2022-12-23:
Why the right wheel does not stop and consumes a current of 1.5A ?

Candas1, 2022-12-23:
Maybe your board has a problem

DmitrySurgut, 2022-12-23:
Maybe, thanks, I'll try on another board

Candas1, 2022-12-23:
Maybe try swapping the wheels to see if the problem is still on the same side

DmitrySurgut, 2022-12-23:
Swapped the wheels, the problem moved from the left wheel to the right wheel. Video - from the left wheel to the right wheel. https://drive.google.com/drive/folders/1LTxlPl3RY7YzgZsmhaCjnE4IPPUD_CdC?usp=share_link

Candas1, 2022-12-23:
As I said this is normal with torque mode without load
But it could be the wheel that stops has more friction

Can you try enabling ELECTRIC_BRAKE_ENABLE and make ELECTRIC_BRAKE_MAX very small? Start with 1 maybe and increase until it's not happening anymore. 
If this works we could maybe make this the default

DmitrySurgut, 2022-12-23:
I lower the accelerator to zero and both wheels should stop, but one wheel still has 2A current and that wheel keeps spinning

Candas1, 2022-12-23:
You are not reading my replies
Bye

closed 2023-01-17 by Candas1