EFeru/hoverboard-firmware-hack-FOC #362: Golf push trolley, low speed high torque,best Approach?

i-windows, 2022-12-30:
### Variant

USART

### Control type

None

### Control mode

None

### Description

I'm going to build a motorized golf push trolley with hoverboard motors and have a couple of questions that I didn't really find answers to in the wiki. I'm not an engineer but sort of understand software.

My cart will be very simple 1 forward speed, can turn left,right and move backwards. To start testing I will use an arduino just to send control commands.

What would like to use is the cruise control, just to maintain a constant speed, but I will also need high torque to be able to climb hills.

What would be the best approach to implement a constant low speed and to also ensure high torque? I can enable the cruise control in the config, but I'm not exactly sure how to set a default speed, nor how to constantly have a high torque. Any tips would be very welcome.

Candas1, 2023-01-01:
Hi,

Cruise control is explained [here](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-HOVERCAR#rocket-features)
[Speed mode](https://github.com/EFeru/hoverboard-firmware-hack-FOC#foc-firmware)  probably the best for your use case

cohnhead66, 2023-01-23:
I have been using this firmware in my golf trolley for some time.  Speed Mode with serial communication from arduino to hoverboard.   You don't need to worry about cruise control.  Just keep sending the command with the same speed.  When you want to turn, just send the command with the speed and the turn coefficient.

I have set up remote operation, happy to share.

i-windows, 2023-01-23:
> I have been using this firmware in my golf trolley for some time. Speed Mode with serial communication from arduino to hoverboard. You don't need to worry about cruise control. Just keep sending the [...]

That's pretty much what I thought, would be great to see what you have done if you don't mind sharing? I just got the remote working yesterday, I have a test cart I have been working on to test out the code. I need to try and attach the wheels to my trolley - this is the hardest part of the build I think :-)

i-windows, 2023-02-05:
> I have been using this firmware in my golf trolley for some time. Speed Mode with serial communication from arduino to hoverboard. You don't need to worry about cruise control. Just keep sending the [...]

any chance you can send your cod on?

cohnhead66, 2023-02-08:
send me your e-mail, i will be happy to share code I have.  

HB Firmware or Remote (Arduino) firmware?

i-windows, 2023-02-09:
cheers, I've setup a repo for you https://github.com/i-windows/cohnhead66
can you dump it there? the remote Arduino would be great thanks

cohnhead66, 2023-05-12:
Were you able to get this working?  Would love to see what you are doing for remote