EFeru/hoverboard-firmware-hack-FOC #363: How to change "flash_write _key" in hovercar mode.

juancargarcia84, 2023-01-09:
### Variant

HOVERCAR

### Control type

FOC

### Control mode

Torque

### Description

I am using the firmware for my son's hovercar. I have strange behaviors with the automatic calibration of the accelerator adc input. It works fine, but it accelerates so hard and the front wheels lift off the ground. By putting the values ​​in config.h "PRI_INPUT2 1, 1000, 0, 4500, 0 ", the boot is very smooth and it was working for a long time. However I wanted to make a few changes to make it speed up a bit faster by setting "PRI_INPUT2 1, 500, 0, 4500, 0". The car accelerated by lifting the wheels again, so I put the previous values, and not after it did not obey the accelerator. After doing an automatic calibration it works very abruptly again, so I thought of erasing the chip completely and then writing the firmware again with the values ​​"PRI_INPUT2 1, 1000, 0, 4500, 0", which previously worked for me, but again it does not obey the throttle until I do an automatic calibration. My only option is to change "write_flash _ key " in config.h, but I don't know how to do it. any ideas? In config.h it literally says this: _"FLASH_WRITE_KEY 0x1107 // Flash memory writing key. Change this key to ignore the input calibrations from the flash memory and use the ones in config.h"_

Candas1, 2023-01-10:
Hi,

If erasing the chip doesn't work, you might have another problem.

Are you sure you haven't reduced the current or speed also?
Are you using same version of the firmware or more recent code?

Candas1, 2023-01-10:
You shouldn't tune the acceleration by changing the calibration.
The auto calibration must be correct.
You can reduce the current and the speed.

closed 2023-01-31 by Candas1