EFeru/hoverboard-firmware-hack-FOC #364: Extract one channel

Kambodia91, 2023-01-15:
### Variant

USART

### Control type

FOC

### Control mode

Torque

### Description

hello I have a hoverboard with two separate bldc control boards, I soldered stm32f103CBT6 48pin on them.  this processor does not have TIM_8.  Is it possible to extract foc control for bldc controller?  The system is equipped with current measurement on two phases, voltage measurement and main current measurement.  I also have a pinout of how the stm32 is connected.  I managed to convert the program so that the engine works only on TIM_1 and works only on SIN_CTRL control in VLT mode.  In the FOC mode, the motor goes crazy and does not respond to the set parameter.  The currents are read into adc_bufor but it doesn't work.  I may have cut out too much of the program.  can anyone help me with Timmer and adc configuration for this STM32f103CBT6 processor.
[image]
[image]

Candas1, 2023-01-15:
Hi,

What was the chip before?
What are the current offsets calculated in bldc. C?

Kambodia91, 2023-01-15:
Previously there was the mm32spin06pf chip.
when I stopped the debugger at the moment.  I had these readings.
[image]

Candas1, 2023-01-16:
Thise are not the offsets.
It's calculated at the beginning of the file

Kambodia91, 2023-01-16:
I added control expressions:
[image]
but sometimes it reads strange current parameters:
[image]

Kambodia91, 2023-01-16:
[image]

Candas1, 2023-01-16:
The offsets look ok
But for the current measurement to be accurate, it has to happen at the right moment, when the low side mosfet is open
I see rlB = 3989, this is not good.
You can use STM studio to visualise the current in realtime
https://www.youtube.com/watch?v=UzvIXeRCZw0

Kambodia91, 2023-01-16:
Adc_buffer:
[image]

Candas1, 2023-01-16:
It's because the current measurements are not good, it shouldn't jump to 4000
I will show you how it should look like when I have time

Kambodia91, 2023-01-16:
ok i set the type to SIN_ctrl and VLT.  I set the PWM to 100;  I started the measurement and stopped the wheel with my hand.  measurements with adc did not change.  it should increase or show something in higher ranges.  i need to work on reading the adc correctly.  and write to the buffer.

Candas1, 2023-01-16:
Is your code on github ?

Kambodia91, 2023-01-16:
managed to upload 😁
https://github.com/Kambodia91/hover-v2-one-bldc-wheel

EFeru, 2023-01-16:
Hi, nice initiative. I also think the issue is in the current reading. Without that FOC cannot function properly.
My suggestion, use a mode that works, SIN is ok, and work out the current measurement, from configuration, to adc calibration and final reading. Good luck.

Candas1, 2023-01-17:
Few tips for STM Studio as I spent a lot of time using it last few weeks:
[image]
- by default STM studio under samples to save resources, display all the data to have a better definition
- acquire only the data used in the viewers, otherwise STM studio pulls the all the data that was imported even if not displayed, which impacts the acquisition rate
- the less data you pull, the better is the definition, so only display one the current of one phase, DC link in your case is not relevant I think as those boards probably don't have the opamp for that

Kambodia91, 2023-01-17:
I have achieved such results.  I had wrong thinking.  I redownloaded clean software.  I changed Tim8 to Tim3 because my chip only has 4 timers.  tim3 for me had similar configurability as tim8.  in the program, these timers affect each other, I don't know exactly how they work, I don't have such knowledge, I'm new to stm.  been doing it for 3 weeks as a hobby.  stm32F103CB has 48 pin so I had to turn off tim3 outputs but the timer works.  similarly to the inputs and outputs that were used in the original design.  Then it was time for adc.  surprisingly adc1 and adc2 started.  😁 it is in multimode.  I couldn't change it so I made some changes to what I have.  when I created an empty project in stmcubeide, in the system clock config there is addclockselection in the empty program I had it set to DIW6.  the original program used DIV4.  in this configuration that I have, I had to use DIV8 after this change my adc and the current measurement calmed down

https://user-images.githubusercontent.com/88535933/212909037-df76e961-1cf1-4d94-9fb0-a3672d931491.mp4

Candas1, 2023-01-17:
Yes on this firmware the ADC frequency is boosted

Kambodia91, 2023-01-17:
please explain how to control the motor with trq foc configuration.  because when I put 30pwm on the input target, the motor accelerates to max speed.  the motor is running without load.

Kambodia91, 2023-01-17:
Foc vlc works ok. PWM controls the speed.  I also checked spd mode.  works best, with resistance it adds power.  thank you for your suggestions. Regards.

Candas1, 2023-01-17:
One thing I thought about recently Emanuel.
A small electric brake by default could prevent this.
It's not really a problem but it would stop the questions 🤣

Kambodia91, 2023-01-17:
https://user-images.githubusercontent.com/88535933/212978366-4bc59a42-6e31-48e9-978c-0cae64d00164.mp4

Power! 😁

Kambodia91, 2023-01-18:
will this command enable I/O pins PB4, PB3, PA15 and I can still program via SWCLK, SWDIO?
AFIO_MAPR_SWJ_CFG_JTAGDISABLE;

Candas1, 2023-01-18:
Check what I did here
https://github.com/Candas1/hoverboard-firmware-hack-FOC/blob/d845bc9bf8f9d854b68301a936e63c075315329f/Src/main.c#L255

I do it only after the button is released

Kambodia91, 2023-01-18:
but what you have in your code disables SWDIO and SWCLK and turns them into inputs.  I need to run the Pb4, Pb3, Pa15 pins because I have a high state on them after starting the processor.  Will the command I gave change the JNRST, JTDO, JTDI pins into outputs?  and SWDIO, SWCLK, stay as they are?

Kambodia91, 2023-01-18:
[image]
I'm interested in this option but I don't know how to implement it so I'm asking

Candas1, 2023-01-18:
I don't know about those pins, you have to try
But that's what I use for extra inputs

Kambodia91, 2023-01-20:
can i use the feedback from the master board (feedback usart2) to control the second board? 
[image]

 the second board would be set to control usart2.  master board sends to feetback ?
[image]

the boards are connected pa2 and pa3, there are pins from usart2 in the documentation, so the configuration is good.  I checked what comes out of the master board and appears in the sendtoslave.start, cmd1, vBat, sumcontrol variables.  There is nothing on the slave board at the input1.cmd input, the same in input2.cmd.

Kambodia91, 2023-01-20:
[image]

Candas1, 2023-01-23:
it's possible
you need to debug

EFeru, 2023-01-23:
I agree Cabdas, a small electric brake would feel more natural to people.

Kambodia91, 2023-01-23:
hey, I managed to use feedback to communicate with the second board via usart2.  works perfect.  but it's time to communicate from the outside 😁.  my cpu doesn't have usart3 but it has usart1.  but when it wants to use usart1 it blocks the adc readings and the wheel starts to go crazy.  I will keep trying.

Kambodia91, 2023-01-25:
Hello.  when I use usart2 FEETBACK to communicate with the second (slave) board and turn on usart1 to control the master board then the program goes crazy the wheel spins as it wants.  but when i disable usart2 and enable usart1 communication, usart1 works ok. why can't i use usart1 and usart2 like this?
[image]

Candas1, 2023-01-26:
Hi,

Your code deviates a lot from the original firmware now, it's hard to help
You need to debug

Kambodia91, 2023-02-06:
hello finally i started usart1 and feedback works
https://user-images.githubusercontent.com/88535933/216865092-623a99cb-f76e-456e-805a-49fdf4391f44.mp4

[image]

https://user-images.githubusercontent.com/88535933/216865732-dd903aac-8d27-4168-a060-b5e3bff9794a.mp4

Candas1, 2023-02-06:
Nice 💪

Kambodia91, 2023-02-24:
hello, i have one problem with uploading the software.  when it makes changes in the program such as disabling or disabling communication or changing the LED pin green red yellow.  I'll just change anything.  After the upgrade, the engine starts to go crazy.  but when i enter debug mode.  I switch lines of code one by one until loop.  then I turn off debug and the processor with the program works fine.  interesting phenomenon.  Can you advise what I can do about this?  I thought the problem might be in build_flags = , in platform.ini.  I remind you that I am using the stm32f103cbt6 processor

Candas1, 2023-02-24:
Maybe that's because you are introducing a delay

Kambodia91, 2023-04-27:
Thanks everyone for help. The program finally works with communication between the boards, but I couldn't deal with the problem of the first launch after uploading the software, I still have to run the program via debbug, I'm currently working on improving security, I added more [date](https://github.com/Kambodia91/Hoverboard-firmware-hack-FOC-MASTER-SLAVE/blob/main/docs/CommunicationDiagram) that exchange between boards. Regards. You can close the topic

Candas1, 2023-04-28:
Have you tried adding delays like I mentioned?

Kambodia91, 2023-04-28:
yes i tried.  with no effect.  I was adding a delay on startup before the loop.  i also tried that add delay between initiation gpio/tim/adc, different combinations.  I noticed a few times during debug when I was on the void powerofmelody() line, it happened after exceeding the Hal_Delay(100) command (they also happened at other times) followed by a transition to Hardfault_handler() in stm32f1xx.c.  then the wheel began to spin without any control.  is it possible that after uploading new firmware, garbage from the previous program remains in the flash memory?

Kambodia91, 2023-04-28:
I put the hall delay command everywhere and success.  after uploading the new firmware, the processor starts ok, then I reduced the number of hall_delay commands to a minimum and I had one right under Intput_init() I must have overlooked this place during the tests.  I've looked more closely at what is in void [input_init()](https://github.com/Kambodia91/Hoverboard-firmware-hack-FOC-MASTER-SLAVE/blob/9ad5806d947042d66e57b8eeba04e381385bb900/Src/util.c#L324).  inside I see initialization first usart2 then usart1, in the original usart2 and usart3 were used 😁 so I switched the order of initialization and HAL_UART_Recive_DMA and success no need hal_delay.

 Candas1 thank you for your advice, without them I would not have done anything myself. As I mentioned, the STM environment is new to me. And what I do is my hobby. My first contact with this processor was in this thread.  Regards

Candas1, 2023-04-28:
💪💪💪 You made it

closed 2023-04-29 by Candas1