EFeru/hoverboard-firmware-hack-FOC #366: Hoverboard does not receive control signal from Web tool.

AndreasDuz, 2023-01-16:
### Variant

USART

### Control type

FOC

### Control mode

Torque

### What is the bug and how to reproduce it ?

Hi there!
I assembled the firmware with USART control (reduced the baud rate to 57600), selected the SIDEBOARD3 protocol, successfully programmed the board, connected the FTDI adapter to the right UART3 channel and tried to set up control via the Web tool. In the "Web tool" settings, I selected the desired COM port, baud rate 57600 and established a connection with the Hoverboard, but the beepper emits three beeps (no UART connection). In the application window, I see that data packets are coming from the board, but I cannot control it.
For diagnostics, I made a test firmware for the Arduino Nano, in it I changed the structure for the Sideboard protocol and specified constant values ​​in the "Send" command. The hoverboard has recognized the packages from the Arduino and is successfully controlling the motors (although I need help figuring out the switch command format).
[image]

Candas1, 2023-01-16:
Hi,
Have you selected the right protocol in the webtool in the control view ?
https://github.com/Candas1/Hoverboard-Web-Serial-Control#control

AndreasDuz, 2023-01-16:
Да, это так. Я пробовал оба варианта: "USART" и "HOVERCAR", но доска не реагирует ни на один, но на дисплее приложения правильно отображается температура доски и напряжение батареи. Адаптер ФТДИ начинает показывать передачу данных сразу же после выбора порта, даже если не выбран экран CTRL. Если бы пакет данных правильно принимался доской, то связь должна была установиться и бипер перестал сигнализировать об ошибке.
Каналы RX и TX UART не перепутаны, приложение перестаёт принимать данные от доски, если переставить провода местами.

AndreasDuz, 2023-01-16:
Yes it is. I've tried both "USART" and "HOVERCAR" but the board doesn't respond to either, but the app's display shows board temperature and battery voltage correctly. The FTDI adapter starts showing data transmission as soon as a port is selected, even if the CTRL screen is not selected. If the data packet was correctly received by the board, then the connection should have been established and the beeper would stop signaling an error.
The RX and TX UART channels are not mixed up, the application stops receiving data from the board if the wires are swapped.

AndreasDuz, 2023-01-16:
Updated data. Even though the FTDI chip is definitely good, the echo-test doesn't work. Since I already know that the RX line is good, I think there is a physical problem on the TX line. I have to deal with this.
Sorry.

AndreasDuz, 2023-01-16:
Once again I apologize, the problem really was in the FTDI chip. Communication with the application improved when I replaced the FTDI with a spare one.
The question can be closed.

P.S.
To avoid open a new topic, can you show with an example how to correctly assemble the control bytes?

0:OFF or 1:ON (for switching input when dual input is used)
Control type 00:FOC 10:Sinusoidal 01:Commutation
Control mode 00:Voltage 10:Speed 01:Torque
Field Weakening 0:off 1:on
00000000000

If I need:
1) No dual input (b0)
2) FOC (b000)
3) Torque (b01000)
4) Field Weakening On (b101000)
5) & 0000000000 (b0000000000101000)
then the final result will be decimal number 40?

Candas1, 2023-01-16:
[image]

closed 2023-01-16 by AndreasDuz

Candas1, 2023-01-17:
👍
Please share what you are doing 😏

AndreasDuz, 2023-01-17:
At the moment I am studying ways to control the motors of an electric car. There was an idea to assemble a light four-wheeled transport with solar panels, for traveling long distances, and to save energy, it is necessary to implement a software differential. According to my idea, the ESP32 module will measure the temperature and actual current consumption of each of the four motors and regulate their load. The driver will send commands and settings to the ESP32, and the response signal will be transmitted via Wi-Fi to the instrument panel.
I do not know how economical differential control will be, but I decided that I must definitely use it.
I just started working on my idea, so unfortunately I don't have photos or videos of the process yet.