EFeru/hoverboard-firmware-hack-FOC #367: Increase parameters r_errInpTgtThres and t_errQual

Firess073, 2023-01-19:
### Variant

PWM

### Control type

FOC

### Control mode

Speed

### Description

How to correctly increase **r_errInpTgtThres** and **t_errQual** parameters in the new version of the firmware?

Candas1, 2023-01-23:
Hi,

Can you give a bit more context why you want to do that ?
What are you calling new version of the firmware ?

Firess073, 2023-01-23:
Hello. 

I built a 4WD platform with two boards from an RC gyroscooter. During its off-road movement, there are moments when the platformer runs into a small obstacle and cannot overcome it and gives error 001 (blocked engine), although the obstacle is not big and the platform should have easily overcome it.
I read in the WIKI that it is necessary to increase the dddddd parameters in order to be more sure that the platform will go off-road without this error.
I use the latest version of the firmware that is available on Git, but I did not understand how to increase these parameters. Tell me, please, how to do it?

Candas1, 2023-01-24:
As mentioned [here](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Diagnostics#1-beep--low-pitch), you need to increase those values in the relevant file, the link point directly to the value, what do you not understand ?
Try doubling value.

Worst case just DIAG_ENA if it bothers you, it's also mentioned in the wiki.

closed 2023-03-01 by Candas1