EFeru/hoverboard-firmware-hack-FOC #368: Multimode in Hovercar Variant wont work

Joe-Holgerson, 2023-01-20:
### Variant

HOVERCAR

### Control type

FOC

### Control mode

Torque

### What is the bug and how to reproduce it ?

I build a two wheel Bobby Car.

In short i have the exact same Problem as this Post: #https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/359

When i use Multimode Control board turns on with powerbutton, but wont run and wont turn off anymore.
When i comment out Multimode Control it all works perfectly.

I couldnt find anything in Troubleshooting concerning this, and the old Thread was closed.
Is there a way i can fix it or is there some problem with the firmware. 
(i can barely programm anything, and cant really contribute, im so sorry, gotta learn)

Candas1, 2023-01-20:
Hi,

https://github.com/EFeru/hoverboard-firmware-hack-FOC/commits/main
This was fixed 12 days ago, that's why the other post was closed

Joe-Holgerson, 2023-01-20:
oh gotcha had an old version... im sorry thanks alot!

closed 2023-01-20 by Joe-Holgerson