EFeru/hoverboard-firmware-hack-FOC #370: E-bike usart and serial Feedback

GioeleM95, 2023-01-22:
### Variant

USART

### Control type

None

### Control mode

None

### Description

Hi! 
Congratulations for the excellent work!  

I have a blocked e-bike project... please help! 
I'm using the VARIANT_USART with Arduino. I use a single motor (// MOTOR_LEFT_ENA) but the serial feedback doesn't seem to work if I use only one wheel, on the serial monitor it shows: "Non-valid data skipped".  
With both wheels, the serial feedback works perfectly.
Can it be the structured of the feedback from the firmware?
Thanks so much!

Candas1, 2023-01-23:
Hi,

This works the same way with 1 or 2 motors.
Data can be skipped if the checksum is not valid, which can happen if the signal is bad.
What arduino are you using ?
I think the initial arduino sketch is using software serial by default but using hardware serial is much better.
https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/efb438bf514c9f8c80471f79ab87d2f7099746ea/Arduino/hoverserial/hoverserial.ino#L10

Candas1, 2023-01-23:
Check the wiki
If you have a ftdi you can try the webtool from a computer
If this works well, you have a problem with your arduino set up

RoboDurden, 2023-03-19:
Arduino Nano and software serial need 19200 baud, 115200 is to much !!
Better go with an esp32 S2 mini :-)

GioeleM95, 2023-03-19:
I used the hardware serial, everything seems to work fine, thanks for the tips.  The software serial didn't make it, this is strange because with another board the software serial worked great 🤷🏻‍♂️.  The only problem is that every time I have to update the Arduino sketch, I will have to disconnect TX and RX... do you know any solution to avoid disconnecting them as the software serial allowed?

RoboDurden, 2023-03-20:
Better go with esp32 as it has three hardware serial ports.
Arduino nano only has one and you have to disconnect when flashing Arduino.
I drive my solar car with Arduino mini and software serial at 19200 baud.