EFeru/hoverboard-firmware-hack-FOC #371: Separate motor speed control & position approximation

skywa04885, 2023-01-24:
### Variant

HOVERBOARD

### Control type

Sinusoidale

### Control mode

Speed

### Description

I want to control both motors separately and change their speeds / read back their position data (sort-off approximated by the HAL sensors in the wheels), before analyzing the entire project, I would like to know if this is possible. In this case I want to control both wheels separately using an Arduino, and check their position (approximation by HAL sensor). Is this possible? Steering won't do the thing, I need full, separate control.

Candas1, 2023-01-24:
Hi, read the wiki

Candas1, 2023-01-24:
Look up tank steering

skywa04885, 2023-01-24:
> Look up tank steering

Okay yes, I've found that. But how can I get the HAL sensor feedback?

Candas1, 2023-01-24:
It not available with this firmware unless you can code
There is a fork doing that it's linked in the wiki

closed 2023-01-31 by Candas1