EFeru/hoverboard-firmware-hack-FOC #374: Motors being erratic and very unpredictable

KuroCrystal, 2023-01-28:
### Variant

PWM

### Control type

FOC

### Control mode

Voltage

### What is the bug and how to reproduce it ?

Ripped out an old Swagway hoverboard and reprogrammed the board successfully. Motor response is extremely erratic and unpredictable and sometimes spins freely with one motor even when there's no input.
video: [https://drive.google.com/file/d/1J4AYCYVpyV4n4ViyDxHjDBdXtg6jO4gl/view?usp=sharing](url)

Candas1, 2023-01-28:
Hi, that's how pwm variant works. Have you checked the wiki?

KuroCrystal, 2023-01-28:
> Hi, that's how pwm variant works. Have you checked the wiki?

Sorry if it doesn't seem that obvious, but it loses control and spins the motor constantly even when the controller joystick is left in the resting position. And sometimes only one motor spins. It's also very laggy/impossible to input anything with the joystick

Candas1, 2023-01-29:
Have you checked the wiki? You are not answering me
What receiver do you have?

KuroCrystal, 2023-01-30:
Nevermind, I solved the issue. No clue how I fixed it however. I have checked the wiki and everything from there is what I followed. I have the FS-ia6b receiver.

closed 2023-01-30 by KuroCrystal

Candas1, 2023-01-30:
Are you sure you transmiter wasn't in ppm mode?
You are wasting your time when you could use ibus with this transmitter/receiver/firmware, it's much more reliable