EFeru/hoverboard-firmware-hack-FOC #375: can i repair this mainboard?

hbnewbie, 2023-01-29:
### Variant

NUNCHUK

### Control type

FOC

### Control mode

Torque

### Description

i have been trying to use my last mainboard for another project but it looks like i have damaged it.  it does have the STM32F103RCT6 chip but the board has less capacitors and uses the motors with an extra white temp sensor wire. 
[image]
 i was able to get it to power on after flashing the alternate board variant option, but cant get it to move it just beeps.  i think i damaged the board by wiring the nunchuk incorrectly, as i thought the google images for pin arrangement were from the open connector side, and not the harness side.  i fixed that issue but am now measuring battery pack voltage at the right sideboard power wire(red) where there should be 15volts(i didnt connect this to nunchuk i used the 3.3 v from programming pin).  i am also seeing batt pack voltage at the power switch pin.  (board still turns on and off normally).  it looks like a voltage regulation problem and am wondering if anyone else has ever fixed a similar issue.

hbnewbie, 2023-01-29:
i dont know if this helps anything, but i noticed this board has a irf9530n at the front and tip127 behind it instead of 2 tip127s like the older boards. battery pack voltage=41.4 volts.  with board powered on at irf9530n i am measuring 30.9volts-gate, 41.4volts-drain, 41.4 volts-source.  at tip 127 i am getting 40.6 volts-gate, 14.5volts-drain, 40.6volts-source.

hbnewbie, 2023-01-29:
sorry for that tip 127 i meant base, collector, emitter instead of gate, drain, source.  dont know what these things mean have just been googling

hbnewbie, 2023-01-29:
ok as a followup i flashed adc variant and its working with the mini joystick.  calibrates and motors move with no warning beeping. my issue might have been with the aftermarket nunchuk i bought.  i saw in this thread where [stdlogicvector] mentions seeing battery pack voltage on RH side cable as well, so maybe its normal on some board variants.  https://github.com/bipropellant/bipropellant-hoverboard-firmware/issues/90

Candas1, 2023-01-30:
Hi,

I created this board_variant based on the feedback of some of the users, but never got to try it.
Thanks for sharing that thread, looks like there are more differences than what I thought.
I might add a warning in the wiki.

I am curious to know if the problem comes from your nunchuk or is this variant just not working on this board.
Need to check if I have such board when I have some free time.

hbnewbie, 2023-01-30:
Thank you for creating that variant.  I will likely order an OEM nunchuk and play with it some more.  I'll report back if I find out anything

hbnewbie, 2023-02-04:
looks like its still a no go.  i tried with a used oem nunchuk but am getting the 4 beep timeout.  with ftdi connected to left cable and using your web serial control, input 1 and 2 always stay at 0.  i have verified 3.3volts and ground at the right cable to nunchuk, and tried swapping the green and blue wires.  i have the two 100k ohm resistors wired as pullup to 3.3v wire.  i am not using the breakout board as recommended in the config file.  i dont know if that could be related. if i pick up another mainboard ill try again to see if its just an issue with this one.

hbnewbie, 2023-02-10:
as an update i picked up another original variant board.  and nunchuk works perfectly.  unfortunately i dont have another of the alternate variant boards to test.  its either something different with that board, or i might have damaged something on it.  will update if i ever get another one.  closing thread

closed 2023-02-10 by hbnewbie

Candas1, 2023-02-11:
Maybe check if the resistors are missing or bad on the right sensor lines

hbnewbie, 2023-02-12:
i have an adapter cable that i made with the 100k resistors wired into it.  i used the same cable on both boards.  on another note, i did find a few issues even when using the original variant board.  the picture under variant nunchuk for the pin arrangement is incorrect.  the scl in the top left and the sda in bottom right positions are swapped. this was the only way it worked for me. basically what this site says. https://domoticx.com/pinout-wii-nunchuk/
i also found that my steering was reversed.  for example when holding right on the joystick the right side wheel would rotate forward while the left side wheel would rotate backwards, instead of the directions it should be going to execute a right turn.  it works fine now after adding the #define INVERT_R_DIRECTION #define INVERT_L_DIRECTION to nunchuk section.  just posting in case others run into these issues.

Candas1, 2023-02-12:
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Parameters#%EF%B8%8F-coefficients

Candas1, 2023-02-12:
The wiki is correct, the pinout is just updide down compared to your link.
Adding a picture like on the link will make it less confusing.

I am talking about the resistors that are on the board itself.

hbnewbie, 2023-02-12:
oh ok i will try to check that out when i have more time.  thanks again

hbnewbie, 2023-02-13:
you were right candas i ended up finding resistor R25 showing over 2000 ohms. i was surprised that it worked when i replaced it since i could barely see it by eye since its so small.  after my crappy soldering job it works normally with nunchuk, but the only difference is after i let go of joystick it freewheels for a little bit, and then jerks hard to a stop.  same nunchuk and config as the variant 0 board.  im guessing something else on the board could be damaged as well. i dont know if ill mess with this one anymore since i already got the other one for cheap.
[image]

Candas1, 2023-02-14:
But the board in the other thread is yet another board type that is not yet supported and that has no dc link current measurement and 2 phases current measurement swapped only on one side.
That's probably not your case.