EFeru/hoverboard-firmware-hack-FOC #376: Motherboard as power inverter for hoverboard wind turbine?

kris86k, 2023-01-29:
### Variant

None

### Control type

None

### Control mode

None

### Description

Hi.
Is possible to use motherboard as accu charger for wind turbine based on hoverboard motor?

ash25mi, 2023-03-17:
Hi,
I'm interested in doing something similar. Did you went on with this project ?

RoboDurden, 2023-03-19:
https://youtu.be/D_PfKFIc_kc

ash25mi, 2023-03-19:
Hello that's cool, thanks for the link !

So if I understand, to have the hoverboard behave as a generator you mostly had to :
- Wait a certain amount of time, or that the motor is spun at a certain speed
- Give an order with a negative speed setpoint ?
I will try to use a hoverboard for a low-tech event where the aim is to build a generator with a bicycle.
If it works with the original 10s battery, next step will be to use a 12v battery to be able to plug a 220V converter.

For use with a wind turbine it would be interesting to implement an MPPT algorithm, so the power generation would be optimal.

RoboDurden, 2023-03-19:
With my pedal crank I can easily generate 1A at 25V. 2A=50W is already exhausting. 100W is the continuous power output of a normal human.
At 12 V the hoverboard motors will have less then 100 W power input or output.

Better find a 24V ac inverter..

Calculate with 14 rpm/V. That way you can find a proper gear ratio.

ash25mi, 2023-03-19:
Thanks a lot for those details !

I will update the results here if we also manage to build the mechanical part.
The simplest way for us will be to roll the hoverboard wheel on the bicycle wheel, so hopefully this will not add too much loss for this demonstrator.

ash25mi, 2023-03-19:
Also to understand, you set a pedaling speed and then it is to you to add an effort to generate power ?
So if you stop pedaling the motor continue to turn ?

RoboDurden, 2023-03-19:
Building a stand for the bike might be more hassle than adding such a freewheel sprocket to a hoverboard motor:

https://youtu.be/ux3u7hCExMk

You could also keep the back wheel when inserting the motor as an pedal assisted bike.

Yes my code detects braking and stops the motor. But if I forget to actively stop pedaling then the motor continues to pedal.

ash25mi, 2023-03-21:
I will see what the mechanical guys are up to for this event. I heard they have a welder, but I suspect we won't be able to do machining. Will see.

RoboDurden, 2023-03-21:
With a 26 inch wheel bike and pedaling 25 kmh you reach 200 rpm.
Simply pushing a 6.5 inch hoverboard wheel onto the 26 inch wheel will make it spin at 200*26/6.5= 800 rpm.
With 14 rpm/V (no load)  you could generate up to 57 volt, quite perfect for a 10s 36 volt battery with max voltage of 42.5 Volts.