EFeru/hoverboard-firmware-hack-FOC #377: Using Hoverboard main board to deliver 100 amps

abyy28y, 2023-02-02:
### Variant

ADC

### Control type

FOC

### Control mode

Torque

### What can we do to make the firmware better?

Hello everyone, good work and big efforts you doing,
Excuse me, I’m using Hoverboard board to make tricycle which can carry heavy objects. The question is, what can I do other than changing mosfets and reinforcing the board with soldering and copper to make it safely deliver up to 100 amps? And which pathway need to be reinforced by copper and solder?
Thanks in advance.

### Describe suggestions or alternatives you have considered

Reinforce the board, changing the mosfets with good one with high amperage capacity

Candas1, 2023-02-12:
you better discuss that in the telegram group that is linked in the repository

RoboDurden, 2023-03-19:
I also already thought if increasing the amps as I drive 1000 Watt scooter motors for my log band saw..

Easiest way is to simply double the MOSFETs by soldering even better MOSFETs on the back of the old ones.

There is a max current limit in config.h which should be increased.
But maybe it would be better to also double the SMD shunt resistors so the higher current will not generate a too high shunt voltage the the op amp would push over the 3.3 volt limit of the MCU !

The foc code derives the phase currents from the MOSFET inner resistance of two phases.
I think this is only for detecting the perfect phase. So only relative currents might be needed.

@Candas1 nice to read that you are working on autodetect. As I have mixed MOSFETs (repair) and mixed motors (a 350+W replacement for a 350W motor) in my setup I am dreaming for a auto configure firmware where all the bldc parameters are  optimized in a closed loop to reach maximum rpm.
Feedback could also be stop and speed to max in shortest time.
The closed loop could simply modify always only one parameter and see if the will bring better performance.

abyy28y, 2023-09-29:
> I also already thought if increasing the amps as I drive 1000 Watt scooter motors for my log band saw..
>
> Easiest way is to simply double the MOSFETs by soldering even better MOSFETs on the back of [...]

thank u very much, and i really appreciate ur help. and may i ask u another question, how can i make the power button works at lower voltage as 12 volt?

abyy28y, 2023-09-29:
@RoboDurden thank u very much, and i really appreciate ur help. and may i ask u another question, how can i make the power button works at lower voltage as 12 volt?

RoboDurden, 2023-09-29:
There is a 30k : 2k voltage divider to reduce the 40V to below 3.3V:

[image]

Try adding another 30k in parallel so a lover voltage will still trigger the onOff input pin.

abyy28y, 2023-09-29:
> There is a 30k : 2k voltage divider to reduce the 40V to below 3.3V:
> 
> [image]
> 
> Try adding another 30k in parallel so a lover voltage will still trigger the onOff input pin.

thank u very much, ill try this.