EFeru/hoverboard-firmware-hack-FOC #379: How can the tx and rx of the right uart be changed to reverse, high and low speed

gutabsgf, 2023-02-08:
### Variant

HOVERCAR

### Control type

FOC

### Control mode

Torque

### Description

How can I change the source code to cancel the double click reversing function of the brake pedal? How to change the source code to change the green line and blue line of the right uart to use the switch to control the reverse function and the high and low speed function

Candas1, 2023-02-12:
You will have to learn coding for that.

gutabsgf, 2023-02-13:
Can you give some tips? For example, which file I need to modify

gutabsgf, 2023-02-13:
> 为此，您必须学习编码。

Can you give some tips? For example, which file I need to modify

Candas1, 2023-02-13:
Have you checked the other issues if the same question was asked ?

gutabsgf, 2023-02-13:
> 如果问同样的问题，您是否检查过其他问题？

Is it okay to give me an email address

gutabsgf, 2023-02-13:
> 如果问同样的问题，您是否检查过其他问题？

I would like to ask you which file needs to be modified to achieve the function I want!

gutabsgf, 2023-02-17:
> 我可能会在今年晚些时候添加类似的东西，但对于哦，我没有时间再见

I'm looking forward to it! Thank you

RoboDurden, 2023-03-19:
Search main.c and setup.c I guess to learn how a digital input pin is initialized.
Then make sure that no uart3 is initialized and initialize the two pins as digital inputs.
Then you could read the two pins in main.c and alter the speed..

gutabsgf, 2023-03-19:
> 搜索main.c和setup.c，我想了解如何初始化数字输入引脚。然后确保没有初始化uart3，并将两个引脚初始化为数字输入。然后你可以读取main.c中的两个引脚并改变速度。
Thank you, can you give me an email?

gutabsgf, 2023-03-19:
Thank you, can you give me an email?

RoboDurden, 2023-03-19:
No. Make progress yourself