EFeru/hoverboard-firmware-hack-FOC #384: requires battery and charger connected to drive the motors after flashing the nunchuk variant

Plant2Pipe, 2023-02-25:
### Variant

NUNCHUK

### Control type

Commutation

### Control mode

Voltage

### What is the bug and how to reproduce it ?

after flashing the board i thought it didnt work cause nothing happened and it started making a faint highpitched sound lik coil whine. i had the power button jumpered for the flash so i reattached the regular power switch and without the jumper the sound is gone. when i press the power button it doesnt latch but while im holding it the led on the board comes on and the sound comes back on but input from the nunchuk still doesnt drive wheels. but if i then connect the charger it starts responding to the controller perfectly but the power switch does nothing and its always on. if i disconnect the battery and connect the charger the led immediately without need to press or hold the power switch  and the coil whine is back but there no drive. strangly the whine is a very slightly different frequency then with just the battery and it also changes freq in response to the controller. i assume that means the stm is powered and working since its responding to i2c and the fact that i can hear it seems to imply its trying to drive the wheels but only with both sources will it succeed. i havent probbed it at all yet cause i dont know what it is id be looking for. i have serial to usb converters in addition to the stlink v2 dongle . i realize this is likely not the code but could be down to my board

Plant2Pipe, 2023-02-25:
fixed it 
found the alternate board code in config that made sense to be the problem and it was. i dont see anything in the main how to about this being a thing so i skimmed right past this setting the first time thru

Candas1, 2023-03-01:
Hi,

It's mentioned in the wiki.
I think this can be closed.