EFeru/hoverboard-firmware-hack-FOC #385: How can I modify the program to support brushed motors?

emo8899, 2023-02-28:
### Variant

None

### Control type

None

### Control mode

None

### Description

I have two 36V brushed motors, how can I modify the program to support brushed motors?"

Candas1, 2023-02-28:
This firmware is for brushless

emo8899, 2023-03-01:
> This firmware is for brushless

I saw someone modify this program to work with a brushed motor by connecting the blue and green wires to the motor. No need to use a hall sensor and the additional yellow motor wire

Candas1, 2023-03-01:
Then you can probably ask him

RoboDurden, 2023-04-04:
It would need a complete rewrite of bldc.c
But I think when connecting the yellow wire to the plus pole of each brushed motor and the blue and green wires to the negative pole of the two motors than at least both motors could be pwm driven (green and blue MOSFETs) independently yet both in the same direction.
But brushed motors have such a bad efficiency that I do not see a good use case for such a firmware.
And as we have to salvage old hoverboards to still get a f103 motherboard with only about 30% success rate, we all have lots of nice BLDC motors piling up :-/

Avt2024Ukr, 2024-11-20:
Ставите режим COM_CTRL та SPD_MODE. Вимикаєте діагностику двигунів. Та підключаєте двигун на дві фази в такому випадку використовуєте плату на 25%. Щоб використати на 100% треба поставити 6-фазний діодний випрямляч.