EFeru/hoverboard-firmware-hack-FOC #386: How can I modify the program to support wheel-motor with 60 degreeses hall sensors? Like ninebot.

novval, 2023-03-01:
### Variant

None

### Control type

None

### Control mode

None

### Description

Hello, Emanuel !!!

How can I modify the program to support wheel-motor with 60 degreeses hall sensors? Like ninebot.

Candas1, 2023-03-01:
Have you checked the wiki ? the firmware is for 120 only

novval, 2023-03-01:
I understand it.
Therefore, I ask if it is possible to change BLDC_Controller or Matlab files, for a new generation of .c-files?

novval, 2023-03-01:
For now, it's still a question. ))
I wanted to understand the principle of how to do it.

Candas1, 2023-03-01:
It's not as simple as that.
We would need such a wheel to test and enhance.
I am actually working on a autodetect feature but I am slow so it's taking time.
And I would need to check with such wheel, I am not sure I have one.

Candas1, 2023-03-01:
So it could work if adjusting the Hall to Position table [here](https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/efb438bf514c9f8c80471f79ab87d2f7099746ea/Src/BLDC_controller_data.c#L150).

The current value is { 0, 2, 0, 1, 4, 3, 5, 0 } which means:
| Hall Position<br>A * 4 +B * 2 + C | A | B | C | Sector | Angle | Comment |
| ------------- | ------------- |------------- |------------- |------------- |------------- |------------- |
| 0 | 0 | 0 | 0 | 0 | N/A | Impossible so triggers Diag error |
| 1 | 0 | 0 | 1 | 2 | 120° |  |
| 2 | 0 | 1 | 0 | 0 | 0° | |
| 3 | 0 | 1 | 1 | 1 | 60° | |
| 4 | 1 | 0 | 0 | 4 | 240° | |
| 5 | 1 | 0 | 1 | 3 | 180° | |
| 6 | 1 | 1 | 0 | 5 | 300° | |
| 7 | 1 | 1 | 1 | 0 | N/A | Impossible so triggers Diag error |
<br>

I had to google this for you as you didn't research or didn't share any information:
[image]

So it looks like one Hall should be inverted, which one should it be ? Have you google it for you wheel ?

Checking the wiki you would know that the hall positions in 60 degree configuration will trigger [motor error](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Diagnostics#1-beep--low-pitch).
I guess if you have tried it you have faced the issue and could have shared this information.

You are not spending any time and creating and issue with just "How can I modify the program", but it takes me 1 hour of my life to investigate it.
I will add this in the wiki, it was shared by a member of the telegram group, [HOW TO ASK SMART QUESTIONS](http://catb.org/~esr/faqs/smart-questions.html)
That's why I am loosing more and more hope with this community, this is not rewarding work at all.
Specially when I am alone because Emanuel lost hope dealing with the same people for years 😂

novval, 2023-03-01:
Thank you so much for your help and patience!!!
I haven't tried connecting a 60 degree wheel motor because I know it won't work.
I searched for a similar conclusion and decided to invert one of the hall sensors.
After the experiment, I will tell you the result.
Thanks again!

gutabsgf, 2023-03-19:
Did you succeed

novval, 2023-05-05:
Hi!
A friend who has such a wheel, for whom I was looking for an answer to this question, said that it did not work out. As a result, he mechanically moved the sensor.
I will try it myself as soon as I have such a wheel.