EFeru/hoverboard-firmware-hack-FOC #390: Voltmeter on motherboard

SchO2n, 2023-03-27:
### Variant

SKATEBOARD

### Control type

FOC

### Control mode

Voltage

### Description

Hello, 
I have a 4WD skateboard project with two motherboards. I would like to add two voltmeters on the motherboards. By connecting the voltmeters directly to the batteries, they stay on all the time. The voltmeters must light up after pressing the push buttons. Could you tell me where can I connect my voltmeters on the motherboards to recover the voltages of the batteries after ignition? Thank you so much. Great project, I'm having fun.

Candas1, 2023-03-28:
Hi, 
Check with a volmeter between the middle pin of the TIP127 and ground
[image]

Why two voltmeters ? Are you going to power each board with a separate battery ?
I think it's better to proceed like [this](https://beta.ivc.no/wiki/index.php/Electric_Bobby_Car_Hoverboard_Upgrade) (Example from a bobbycar)

SchO2n, 2023-03-28:
Hello, I will test on the center pin of the TIP127. Thank you for your valuable advice.

Candas1, 2023-03-28:
Don't short anything...

4wdgif, 2023-04-29:
Don't short circuit anything and take ya time and always triple check your work

Candas1, 2023-05-04:
Hi, any news ?

SchO2n, 2023-05-05:
Hello Candas1, 
thank you for taking news of my project. On one of the two motherboards, I had two cables (red and black) coming out of the card at the places you told me. So I tested and I seem to be able to get the voltage back using these cables. On the other motherboard, I plugged in as you told me and I was able to recover the voltage. Everything was working fine, and after several ignitions, the motherboard sparked and it wouldn't turn on anymore. The flash still works but the motherboard does not turn on anymore. I read that it was certainly the mofset. But I don't know how to test them. I found another card and there are already the two cables like on the first motherboard. I still have to wire everything up.
Tks

Candas1, 2023-05-05:
OK cool.
You are talking about the wires for the Bluetooth\speaker board, I forgot about that, good point.
We can close this then.

closed 2023-05-05 by Candas1

Candas1, 2023-05-05:
I added it [here](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Battery#voltmeter)