EFeru/hoverboard-firmware-hack-FOC #393: problem gd32f103

chachinoi33, 2023-04-08:
### Variant

IBUS

### Control type

FOC

### Control mode

None

### Description

hello, I have already done a project in ibus flysky with an stm32f103rct6 card and it worked very well, I now have a gd32f103ct6 card and I encounter this problem. https://youtu.be/rMqWm0DB0JI
Do you know what the problem is?
THANKS

Candas1, 2023-04-08:
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Firmware-Compatibility#toolbox-troubleshooting

chachinoi33, 2023-04-08:
Thank you, I just changed the direction of the right motor phases and it is now turning, but they are still idling

chachinoi33, 2023-04-08:
I switched to pwm, everything works but I have the left motor which has no more torque (I can stop it by hand)
I tried in torque but it's not better, do you have any idea?

Candas1, 2023-04-09:
Ibus shouldn't be the problem, it's better than pwm.
It's still a hall problem.

chachinoi33, 2023-04-09:
thank you very much for the support!  I will try tomorrow to move the hall sensors

chachinoi33, 2023-05-06:
hello, sorry I did not give news.  I managed to get everything working by inverting 2 hall sensors, but ended up ditching the hoverboard motherboard for ebike controllers.  https://youtu.be/wSHKioaHZvM

Candas1, 2023-05-06:
👍

closed 2023-05-06 by Candas1