EFeru/hoverboard-firmware-hack-FOC #394: Acceleration stuck after throttle release

lfbferraz, 2023-04-10:
### Variant

ADC

### Control type

Commutation

### Control mode

Voltage

### What is the bug and how to reproduce it ?

Hey guys,

I apologize for writing in the bugs section, but I couldn't find an appropriate section to share my issues with you experts.

I modified my daughters Ride on Car, removed the cheap 25w motors with plastic gearbox to the hoverboard chassis with the motors fitted to the original plastic wheels and a few more modifications.

I'm using a RC tx/rx connected to the right PWM cable, using only the speed input, and I've changed the steering coefficient to 0 to ignore any disturbance from the open steering channel.

The car runs fine when the wheels are off the ground or unloaded.
The problem is that when my daughter is in the car, I press the accelerator lightly and the car accelerates hard, and after I release the accelerator it continues accelerating for a few seconds. I almost crashed the car with my daughter in it and it's very dangerous.

Thing I tested:
Change voltage to torque - does the same
Change RATE, FILTERS, COEFFICIENTS, I_*MAX and N_MOT_MAX, also no improvements.

I have to say it was running fine and issue started after:
- I changed 8'' wheels from a Hoverboard XL to 6'' wheels from a common hoverboard
- I connected the RC receiver at 5v through a DC-DC (15 to 5v) on the same cable as the right board cable, to use a steering servo, which is very weak at 3v. (the right cable in theory supports 5v, and in my mind I wonder what difference it makes if the no-load operation is fine?)
- I put a kill switch and fuse between the battery and the mainboard (I'm glad I did this before I let my daughter ride)

I don't actually have a lot of time to dedicate to this project, so I wanted to ask if anyone has any clues as to what might be wrong?

The next thing I plan to test is remove the steering servo and power the RC receiver back to 3v. Also build a new code with a 50 rpm cap and a very low rate for safety).

Thanks in advance
Luis

Candas1, 2023-04-10:
Hi,

Have you checked [this](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-PWM)?

The pwm implementation is prone to noise, it needs rework.
But there are advices on the wiki page to reduce the noise.

lfbferraz, 2023-04-10:
Hi Candas1

Yes, I did red, but to be honest didn't care enought as it was previouly working.
I've just twisted the PWM and GND cables together and kept it short, around 5cm, but still connected to the board connector.
I will try to solder a capacitor and make it even shorter, as I can't find a shielded cable or ferrite around.

Thanks

Candas1, 2023-04-11:
Maybe you can also try another variant to prove if the problem is the pwm signal.

lfbferraz, 2023-05-04:
Yes and no.
I manage to change the transmitter to board cable to a shorter one, twisted
with the gnd and soldered a capacitor on the board input.
But didn't had time to test yet.

Hope to test it next week.