EFeru/hoverboard-firmware-hack-FOC #396: Is the firmware compatible with the GD32F303

tamano123, 2023-04-20:
### Variant

USART

### Control type

FOC

### Control mode

Speed

### Description

Hello, i want to buy a motherboard and i found one (chinese) that run with the GD32F303 micro-controller. I want to know if this micro-controller is compatible with the firmware ? Thank you

Candas1, 2023-04-20:
Hi,

Check the wiki and use the search button to find similar questions.

tamano123, 2023-04-20:
Thank you @Candas1 for your answer but i didn't find a solution (or may be i am searching in the wrong place :-)) in the wiki (https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Firmware-Compatibility) about the compatibility of the GD32F303 microcontoller. I also made a search in the issues (already closed and open) but didn't find any result... please advice me

Candas1, 2023-04-20:
So you are not able to use the search
https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues?q=F303

tamano123, 2023-04-20:
So sorry. Thank you -:)

closed 2023-04-20 by tamano123