EFeru/hoverboard-firmware-hack-FOC #397: motherboard voltage

hbnewbie, 2023-04-21:
### Variant

None

### Control type

None

### Control mode

None

### Description

does anyone know a part number that can be used to upgrade the dc to dc converter?  i have seen gerkon mention this in this thread: https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/87 . i am assuming he is referring to the part labeled RS3M but i havent been able to find a part listed as an upgrade for it on google.  i have hit a max of 40 mph with a 14s lawnmower battery in sinusoidal mode, but am curious how high in voltage i can go before the motherboard blows.  i would be upgrading the caps on it and i have spare parts when they fail. 
[image]

cohnhead66, 2023-05-12:
What tires are you using?  Solid or tube tires?

hbnewbie, 2023-06-22:
theyre just cheap tires with tubes.  i ended up deciding not to try any faster for safety reasons.  it did have a fair amount of wobble at top speed and i didnt want to upgrade to more expensive/legit gokart parts.