EFeru/hoverboard-firmware-hack-FOC #398: Hoverboard model database (Jetson Sphere)

robcazzaro, 2023-04-23:
### Variant

HOVERBOARD

### Control type

FOC

### Control mode

None

### What can we do to make the firmware better?

I just discovered this project, and bought a used hoverboard off of Craigslist (Jetson Sphere). Upon opening it, I discovered it has a split board with an unsupported processor (MM32SPIN05PFOP). Which means I can only use the motors and battery (and it's ok, that's what I need anyway).

But I was thinking that creating a database of known models (supported/unsupported) could benefit everyone. Ideally, I would have bought a unit with a controller that could be hacked.

I'm using the opportunity to enter the Jetson Sphere data:

Battery/motors 36V
Split board with a MM32SPIN05PFOP each
Unsupported by any hacked firmware and unlikely to be supported by a port due to the processor used

The MM32 line is an M0 core, pretty well documented and seems to have decent software and HAL support, I'm enclosing the documentation which can be found here https://www.mindmotion.com.cn/products/mm32mcu/mm32spin/mm32spin_specific_mcu/mm32spin05/
Alas, there's no available dev board for that specific variant (the MM32SPIN05PF**OP** variant adds opamps to the generic MM32SPIN05PF, which is available albeit expensive)

The MM32SPIN05PFOP is almost pin compatible with the STM32F103C8T6, apart from pin 35 and 36, which are PD2 and PD3 on the MM32SPIN05PFOP and VDD_2 and VSS_2. So it *might* be possible to remove the MM32SPIN05PFOP and install an STM32F103 (need to check if pins 35 and 36 are connected), but even if I'm confident I could replace the chip (bismuth and hot air), it would still require building a board schematic to figure out all the connections)

Thanks for sharing this project, I learned a lot!

[DS_MM32SPIN05x_q_EN.pdf](https://github.com/EFeru/hoverboard-firmware-hack-FOC/files/11304897/DS_MM32SPIN05x_q_EN.pdf)
[UM_MM32SPIN05x_q_EN.pdf](https://github.com/EFeru/hoverboard-firmware-hack-FOC/files/11304896/UM_MM32SPIN05x_q_EN.pdf)

### Describe suggestions or alternatives you have considered

None

Candas1, 2023-04-24:
Hi, thanks for sharing

[This](https://github.com/Kambodia91/Hoverboard-firmware-hack-FOC-MASTER-SLAVE)  guy actually swapped the mm32 with a stm32 and ported the foc firmware

robcazzaro, 2023-04-24:
Thanks @Candas1. I actually did see that repository while searching, His original processor (MM32SPIN06PF) is similar to mine, but the boards are different, requiring some work to ensure that the schematic is understood. Considering the cost of an original STM32F103 (~$8) and the work involved, not sure it's something I want to do (also, it's a LQFN48 package and with more limited features compared to the one used in this repository).

More than anything, I thought that having a hoverboard database could help people looking for used hoverboards on Craigslist or thrift shops in selecting the compatible ones (nothing is guaranteed with Chinese device, identical looking ones can have different boards, but it's a start)

reed-dan, 2024-01-06:
[RoboDurden](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/)has a good repository of boards going. I recommend tracking/documenting the frame material & wheel material also. I have noticed a lot of the Jetson models are constructed on plastic  (which tends to crack) unlike some other models with Aluminium frames.

I believe you likely have this board
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/22
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/tree/main/Schematics_2.8

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/