EFeru/hoverboard-firmware-hack-FOC #403: Source for compatible mainboards?

ShawnLilly, 2023-05-08:
### Variant

None

### Control type

None

### Control mode

None

### Description

I’ve been looking all over the internet but I can’t for the life of me find any mainboards available that are compatible. Are they even sold anymore or is the only option salvaging older used hoverboards and crossing your fingers?  

I found a [Link](https://t.me/joinchat/BHWO_RKu2LT5ZxEkvUB8uw)  in a similar question for a group that might have some to sell That link appears invalid now though.

Candas1, 2023-05-08:
Hi,

No the link is not invalid it works for me.

ShawnLilly, 2023-05-08:
Oh dang I guess it just doesn’t work on safari :). Thanks for letting me know. I’ll check out the group and report back.

ShawnLilly, 2023-05-10:
Well it looks like buying used and hoping is the only real option I could find.

ShawnLilly, 2023-05-13:
We’ll I found [this](US $36.07  18％ Off | Circuit Board Hoverboard Motherboard Mainboard Control Taotao PCB for 6.5/8/10" 2 Wheel Self Balance Electric Scooter Skateboard https://a.aliexpress.com/_m0HYmkY) on aliexpress. The person i messaged said it has a STM32F103RCT6. I ordered one but I’ll update when I get it.