EFeru/hoverboard-firmware-hack-FOC #405: black yst board varant with gd32f103

Plant2Pipe, 2023-05-14:
### Variant

None

### Control type

None

### Control mode

None

### Description

i have a black mainboard yst-6.5"/10"-pcba-10. the info says these yst aren't compatible but they r talking about a similar looking board with an at32...    since this board has the micro that's compatible is it just a new pinout or is it still incompatible

Candas1, 2023-05-15:
Hi,

You mean this ?
https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/230

Plant2Pipe, 2023-05-18:
> Hi,
> 
> You mean this ? #230

no. it looks lik that and has yst on one side but directly across from that where the other one has nothing this one say "yst-6.5"/10"-pcba-10" and it has a gd32f103 mcu and ive successfully unlocked it and uploaded this repository's code but it doesnt turn on.
some other differences r that it has a smaller 3 pin connector between the switch connecter and the charge connecter and that its hall connector is a 5 pin connector plus a 2 pin connector with just one white wire populated, it also has discreet gate drive circuitry and different voltage regulation

Plant2Pipe, 2023-05-18:
this is it halfway down this page https://github.com/lucysrausch/hoverboard-firmware-hack/issues/32
[image]

Candas1, 2023-05-18:
Have you checked the compatibility page in the wiki?

Plant2Pipe, 2023-05-19:
> Have you checked the compatibility page in the wiki?

yes and it says boards that hav this mcu should be compatible but there r no pictures of this variant of the board.
but through experimentation ive got it almost fully working. now the power latch and piezoelectric r working for sure and im assuming the rest but since im using the nunchuk variant i need to find the right (usart/uart?) wires.
my version has a 12 pin header and a 9 pin which split into a 7pin that is short that must hav went the the near side sensor board plus the 5 pin hall connector.    and the far side 5 pin hall connector and a long 4 wire connector respectively.
im assuming the wires i need r on the 7 pin connector but the wire colors r messed up with one of the two black wires being 15v and one of the two reds being ground . the red and black and yellow wires that arent populated on other boards r labeled 4DA0, 8A"0/D", 4AM

Candas1, 2023-05-19:
Can you share a picture of the board?