EFeru/hoverboard-firmware-hack-FOC #406: Wrong speed control when left wheel goes forward

ombra80, 2023-05-14:
### Variant

PWM

### Control type

FOC

### Control mode

Speed

### What is the bug and how to reproduce it ?

The left wheel is ok, and the right wheel too (when it spins backwards). when the left wheel spins forward, as soon as it feels a little load, it spins in the crazy opposite direction. it seems that the algorithm that controls the forward/left speed works in reverse.
How can i fix it?
...congratulations, really a great job!

Candas1, 2023-05-15:
Hi,

have you checked the wiki ?

ombra80, 2023-05-27:
Yes, I checked it. In part, I've solved it with full chip erase. However, when the weehls spin forward, they work perfectly. When they spin backwards, they have very small torque. Any suggestions to fix it? Thank you in  advance!

Best
Fabio

Candas1, 2023-05-28:
Can the wiring of the phases/halls.
What type of wheels are you using?

ombra80, 2023-05-28:
The wheels are 10 inches, I've tried all combinations of the hall sensors, but the correct one is the first one.
if I reverse the direction( #INVERT_R_DIRECTION
#INVERT_L_DIRECTION)
the problem moves in the forward direction) I doubt it's a hardware problem.

in serial debugging, when I command maximum speed backwards with wheels in the air I see:

in1: -1000
cmdL -500
cmdR -500

if I brake the wheels I see:
in1: -1000
cmdL -500 /0/-500/0/-500
cmdR -500/0/-500/0/-500
🤔

Candas1, 2023-05-28:
Pwm variant picks a lot of noise, that's why I told you to check the wiki. It tells you how to deal about that

ombra80, 2023-06-04:
it wasn't a noise problem, I solved it with short and shielded cables.
I replaced the board and now it works fine.
it was probably defective or not 100% compatible.
now work! https://youtu.be/uB3rUG6iHAE