EFeru/hoverboard-firmware-hack-FOC #408: 60 degree hall sensors

humble800, 2023-05-20:
### Variant

USART

### Control type

FOC

### Control mode

Torque

### Description

Hi @Candas1, Thank you and @EFeru for developing and maintaining the FOC software.  It has been working perfectly on my 6” hoverboard motor that has 120 degree hall sensors with exact internal wiring as here: https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/main/docs/pictures/motor_inside.jpg.  I am working on an 8” hub motor with 60 degree hall sensor and will appreciate some tips to make the software work with that motor.

I got an 8” motor from an old Halo hoverboard. This 8” wheel has 60 degree hall sensors.  It also has different hall sensor spacing and placement (1.5 poles in between) vs. the 6” wheel that has hall sensor spacing of 3 poles.  In addition, the sensor wire orders are also different.  This 8” wheel has green hall sensor goes first, followed by the yellow hall sensor wire (which is reversed to produce 60 degree), and lastly the blue hall sensor wire.  This is different from the Yellow->Blue->Green order of the working 6” wheel.  I believe both wheels use WYE wiring.  Please see pictures below.

I have tried inverting the hall value as @Candas1 noted [here](https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/efb438bf514c9f8c80471f79ab87d2f7099746ea/Src/bldc.c#L174) and [here](https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/efb438bf514c9f8c80471f79ab87d2f7099746ea/Src/bldc.c#L212).  I also connected the 8” hall sensor wire in different orders to match with the order of the working 6” wheel.  In particular I connected the green hall wire to yellow (Hall C as shown here https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/main/docs/pictures/mainboard_pinout.png), yellow to blue (Hall A), and Blue to Green (Hall B).  I haven’t been able to make it work so far.  I have tried many combinations of matching different hall sensor wires, inverting one of the three hall values, and even tried different combinations of phase wires.

The next step I am thinking to try is to adjust the Hall to Position table [here](https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/efb438bf514c9f8c80471f79ab87d2f7099746ea/Src/BLDC_controller_data.c#L150) as @Candas1 suggested.  I couldn’t figure out the meaning of sector and angle: https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/386#issuecomment-1450640574.  Could you give me some suggestions please?  Given that the physical placement of the hall sensor spacing on the 8” wheel differs from the working 6” wheel, would it even possible to make the FOC software work this 8” wheel with 60 degree hall sensors?

Thank you very much!
[image]
[image]

Candas1, 2023-05-20:
Hi,

It's the first time I see that, it doesn't look like the middle hall sensor is inverted. 
What do you mean by not working? 

I have a very experimental firmware with some kind of autodetect. It's not finished yet but I will check if I can publish a version that can at least give the values for the hall to position table.

Candas1, 2023-05-20:
I also see there are 7 coil wires in parallel?
There is an extra white wire for temperature, you have seen that you sure use the alternative board variant?

Can you share pictures of the other side of the wheel and the hoverboard? I want to check if I have anything like this

Candas1, 2023-05-20:
Can you try this [fork](https://github.com/Candas1/hoverboard-firmware-hack-FOC/tree/ADC-test) ?
It will make the wheels turn and output some information on right USART.

It's set to work with [this](https://github.com/Candas1/Hoverboard-Web-Serial-Control#protocol---hovercar) for my test.
Use it only for diagnostics, I made a lot more changes that I still need to test.

humble800, 2023-05-20:
Many thanks, @Candas1.   Sorry it took me a while to set up the test environment.  I believe I got the output message you are looking for.

I built VARIANT_USART variant (platformio.ini); changed BOARD_VARIANT to 1 (config.h), changed #define VBAT_PIN GPIO_PIN_1 (define.h to fix the compile error for VBAT_PIN).

I only connected the left motor.  I did swap the yellow and blue color hall sensor wire from the motor to the HB mainboard.

I did two tests below:
Test #1. I left the phase wire unchanged (meaning matching the same phase wire color from the HB with the motor).
Test #2. Swapped the Blue and Yellow phase wire between the HB mainboard and the motor.

The output message from UART3 are listed below.  Thank you again!!

/***********************************
**Test #1**
***********************************/
Using the configuration from config.h
a mid-resting potLimits Input1: TYP:0 MIN:-1000 MID:0 MAX:1000
Limits Input2: TYP:2 MIN:-1000 MID:0 MAX:1000
-- Motors enabled --
Warning, Serial Timeout
***repeat timeout msg***
Warning, Serial Timeout

 Right  Motor
+-------+-----+------+------+-----+-----+
|-------|DCCUR|PHA-AB|PHA-BC|DCCUR|PHASE|
+-------+-----+------+------+-----+-----+
|Offsets|01966|002015|002006|_____|_____|
|Phase A|-0003|000017|000009|___OK|_____|
|Phase B|-0003|000012|000009|___OK|___OK|
|Phase C|-0002|000007|000004|___OK|__NOK|

Hall Positions
3,:,0,1,4,3,5,0,
Hall Angles
179,0,0,0,0,0,0,0,

Hall A  B  C
Min- 0 0 0
Max- 0 0 0
OK?- 0 0 0

 Left  Motor
+-------+-----+------+------+-----+-----+
|-------|DCCUR|PHA-AB|PHA-BC|DCCUR|PHASE|
+-------+-----+------+------+-----+-----+
|Offsets|01965|002019|002027|_____|_____|
|Phase A|-0008|000035|000010|___OK|___OK|
|Phase B|00000|000016|000041|__NOK|___OK|
|Phase C|00001|000000|-00014|__NOK|_____|

Hall Positions
0,1,3,2,5,0,4,0,
Ha�l Angles
0,33,139,102,277,310,193,0,

Hall A  B  C
Min- 0 0 0
Max- 1 1 1
OK?- 1 1 1
Warning, motos right error code 3
Warning, motor right error code 3
Warning, motor right error code 3
Warning, motor right error code 3
Warning, motor right error code 3

/***********************************
**Test #2**
***********************************/
Right  Motor
+-------+-----+------+------+-----+-----+
|-------|DCCUR|PHA-AB|PHA-BC|DCCUR|PHASE|
+-------+-----+------+------+-----+-----+
|Offsets|01964|002014|002004|_____|_____|
|Phase A|-0003|000012|000003|___OK|_____|
|Phase B|00000|000011|000008|__NOK|___OK|
|Phase C|00004|000015|000008|__NOK|__NOK|

Hall Positions
3,2,0,1,4,3,5,0,
Hall Angles
179,0,0,0,0,0,0,0,

Hall A  B  C
Min- 0 0 0
Max- 0 0 0
OK?- 0 0 0

 Left  Motor
+-------+-----+------+------+-----+-----+
|-------|DCCUR|PHA-AB|PHA-BC|DCCUR|PHASE|
+-------+-----+------+------+-----+-----+
|Offsets|01965|002019|002025|_____|_____|
|Phase A|-0005|000029|-00048|___OK|___OK|
|Phase B|00000|-00017|000025|__NOK|___OK|
|Phase C|-0002|-00023|-00048|___OK|_____|

Hall Positions
0,3,1,2,5,4,0,0,
Hall Angles
0,127,57,91,284,202,330,0,

Hall A  B  C
Min- 0 0 0
Max- 1 1 1
OK?- 1 1 1
Warning, motor right error code 3

Candas1, 2023-05-20:
ok there are some weird values but it's probably my code that is buggy.
So what if you use now the hall positions from debug output [here](https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/efb438bf514c9f8c80471f79ab87d2f7099746ea/Src/BLDC_controller_data.c#L150) with the main firmware ?
0,1,3,2,5,0,4,0, or 0,3,1,2,5,4,0,0, depending on how it's wired now.
If the hall sensors were really spaced at 60 degrees I would expect the 1 or last value to be non zero based on the theory. Unfortunately I don't own sure motor.

humble800, 2023-05-21:
Thank you again, @Candas1.  I tried to overwrite the hall positions from debug output with the main firmware; unfortunately no luck yet.  For what it is worth, I did notice change the phase wire would change the hall position readings from the debug window using your code.  So, I updated the code and test again without changing the phase wire connection until the hall position readings remains the same.   I did that for the 6 permutation of the phase wire connections.  So far no luck.

1. Could you tell me if the hall labels on this picture (https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/main/docs/pictures/mainboard_pinout.png) matches with the phase wire color scheme here?  
2. Does it also match the A, B, C label noted on the hall position table here: https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/386#issuecomment-1450640574?  
3. Additionally, does A mean u, B mean v, and C mean w in the main firmware?

Thank you again!

humble800, 2023-05-21:
I take it back! I did the following it seems working except the wheel is not going the correct direction.  Meaning, the left wheel rotates backwards (counterclockwise instead of clockwise).

Here are what I did: 
1. swap the yellow and blue hall sensor wires as I showed on the video before.
2. change the hall positions as { 0,1,3,2,5,0,4,0 } based on the debug output based on @Candas1's code
3. no change in the phase wire connections.

Right now the wheel is rotating the opposite direction as expected.  So I suspect changing the hall position array could solve this problem; but I don't know how.  Some guidance will be appreciated!

Candas1, 2023-05-21:
You can invert a motor in config.ini. It's described in the wiki.
But wouldn't this have worked by swapping phases and finding the right combination ?

But it's good news, this autodetect works. I just need to run it on demand and save the results to eeprom

humble800, 2023-05-21:
Thank you, @Candas1.  I will give it a try next weekend.  Could you please confirm if the statements listed here are all true? https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/408#issuecomment-1556065209  I am thinking eventually after I find the correct wire mapping, I may be able to modify the firmware by redefining the pins on define.h and make it work.

humble800, 2023-05-21:
I mistakenly closed the request.  Your comments to the prior questions are appreciated.

closed 2023-05-22 by humble800