EFeru/hoverboard-firmware-hack-FOC #409: Rt side board not working

abyy28y, 2023-05-23:
### Variant

HOVERCAR

### Control type

FOC

### Control mode

Torque

### What is the bug and how to reproduce it ?

Hello everyone, good work you are doing and I wish more success for you.
 I flashed my board as mentioned in wiki. I’m using 2x pedal. Every thing is ok except, when I turn it on 
1- battery indicator lamps blinks only once.
2- photosensors didn’t work, I couldn’t change the controlling mode or turning on fw
3- when I press the photo sensor back led is on but with no changes regarding FW or controlling mode!!!

abyy28y, 2023-05-23:
Hi candas1, I have stm32f103 on my board, yes I use the right firmware downloaded from GitHub, modified by vs code and flashed to board by st link utility prog.

Candas1, 2023-05-23:
You need to make sure the TX works on the sideboard and RX works on the mainboard

abyy28y, 2023-05-23:
I uncomment sideboard serial usart3 and feedback serial usart3
Any something else?

Candas1, 2023-05-23:
I mean the hardware can be faulty

abyy28y, 2023-05-23:
Ah, I have large number of side board in working state, I tried different one with no change

Candas1, 2023-05-23:
> I uncomment sideboard serial usart3 and feedback serial usart3 Any something else?

You shouldn't have to change anything with hovercar variant

abyy28y, 2023-05-31:
Sorry for late still nothing, wondering how to debug the board while I’m using rt cable?