EFeru/hoverboard-firmware-hack-FOC #410: ADC Current

Kambodia91, 2023-05-26:
### Variant

USART

### Control type

FOC

### Control mode

None

### Description

Hello.  please explain to me how the current measurement on the motor phase is done.  I found in Wiki (Used to measure phase current for FOC through low-side Mosfet RDSon as shunt resistor) it's about how we know if 1A measured for the motor corresponds to 1A in the program.  Is it somehow calibrated?  I have other motherboards and I think something is wrong with it.  because I_MOT MAX current is set to 30A and it cuts off the wheel further.  I also thought that the security system on the emergence stop board worked and turned off the timmer, so I turned it off for testing and still the same.

Candas1, 2023-05-26:
I had captured [this](https://docs.google.com/presentation/d/1pdJqmeX-wLChsHSGvrFmTE76WcokVxpkw2e2r_Hd5YY/edit#slide=id.p) some time ago.
I need to review it again but it might help.

Candas1, 2023-05-26:
I am also listing there a lot of improvements I am working on now.

Kambodia91, 2023-05-26:
I reviewed what you sent and I came to the conclusion that I need to find resistors at Opamp that set the gain and locate the shunt resistor and find out what mosfte is there to know what Rdson has to correctly calculate A2BIT_CONV.

Candas1, 2023-05-27:
You could also try [This](https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/408#issuecomment-1555974415)
I will power each phase separately at startup and output the measurements.
But be careful it's experimental

Kambodia91, 2023-05-28:
Unfortunately, I can't upload this firmware due to modifications for a smaller processor.

I traced the paths on my board and the resistor values are the same as here:     :)
 
[image]

[image]

Mosfets on the board are [SW076R68E7T](http://www.semipower.com.cn/uploadfile/cp/mosfet/SW076R68E7T.pdf)

maybe my wheel is too weak?

Candas1, 2023-05-28:
I don't think the limit is very accurate in any case.
You say when you set 30a you reach much less?
Have you checked the wiki section about this parameter?

Kambodia91, 2023-05-28:
on Wednesday I will have a clamp meter to check what current is coming from the battery and what is on the phase.  maybe there are those 30a.  the second thing I need to check is whether the BMS of the battery does not disconnect the electronics.  
Photo of the BMS from the damaged original hoverboard battery that I use: 
[image]
the [video](https://youtu.be/i2_hToZfvqY) shows how the controller gets an error when trying to start.

Candas1, 2023-05-28:
Check in the wiki what to do with single beep.

Kambodia91, 2023-05-28:
in my case, the motor locks up under too much load.  these t_errQual values, as I understand it, are the time after which an error will be triggered.  r_errInpTgtThres is the hall input error when blocking the wheel.  how much can i increase these values?

Candas1, 2023-06-01:
Just double it
You can also disable diagnostics if it bothers you

Kambodia91, 2023-06-01:
I measured the current.  The battery cable was max 15A per channel.  Must have built-in current limiter in BMS.  As for the parameters in BLDC_controller_data.c, I increased the parameters as described in the Wiki.  the controller does not get into an error but now it turns off my batteries, the BMS protection (temperature) works.

Kambodia91, 2023-07-10:
Is it possible to change the PID characteristics so that the wheel starts with maximum power and the set speed, so far it gradually increases power when it encounters resistance, The problem is that with the size of my wheels about 30cm, they do not have the strength to start spinning / starting. I'm trying to achieve movement like a tank, turning on the spot. can you suggest something?
the wheels are said to be 150W, it's less than aluminum 250W, maybe there is a reason here?

closed 2023-12-13 by Kambodia91