EFeru/hoverboard-firmware-hack-FOC #412: multimode and baby car

HenryNickel, 2023-06-05:
### Variant

HOVERCAR

### Control type

FOC

### Control mode

Torque

### Description

I programmed the board in the hovercar version. Everything works great, thank you for a great project using old hoverboards. My problem.
I don't know if I set something wrong, or if it is wrong preset, but the multimode is not working well. In 1st gear the baby car won't start, just when lifted the wheels turn very slowly but it has no power. In 2nd speed it goes about 4 km/h and in 3rd speed it goes too fast again and is dangerous for the baby. What am I doing wrong?
I've set the inputs to autocalibrate. I was wondering if I had unconsciously thrown the current and speed limits. But after about 5s of holding the button after the beep, I just set the pedals and didn't immediately press the button again that I would get to those limit parameters. Does the preset multimode work for you? 
I'm also wondering if I should have left VOLTAGE MODE in the Hovercar settings sector. My setting is TRQ_MODE:
#ifdef VARIANT_HOVERCAR
  #define FLASH_WRITE_KEY 0x1107  
  #undef CTRL_MOD_REQ
  #define CTRL_MOD_REQ TRQ_MODE
  #define CONTROL_ADC 0

Candas1, 2023-06-05:
Erase the chip and flash again to make sure nothing is saved in eeprom.

IerwinDE, 2023-08-21:
I think I'm haing the same issue in my 4wd kart with multi mode.
Mode 1 and 2 are to weak to even get it moving and mode 3 is dangerous for inexpereinced drivers.
I will perform some tests and report back with my findings

HenryNickel, 2023-09-02:
Here is my tested setup, fully satisfactory. Since the discussion here is almost dead, I had to figure it out myself.
My wheels are 10 inch. On narrow 10x2.125 tires, top speed was 27 MPH.
On the wider 10x3.0, top speed is 25 MPH. The kid's 25 kg.
When turned on without hitting the brake or throttle, the middle speed is set at 14 km/h.
When switched on with the brake on, the slowest speed can be set at 7 km/h.
When switched on with both brake and throttle (only used on wide in-line track) syn reaches 25 km/h.
I have activated his "handbrake", which allows him to stop the car on a hill and get off. 
This feature, once the car is stationary, holds the car in place and if you try to push it in any direction, the motors push against you.
#define STANDSTILL_HOLD_ENABLE

Here's my setup:

#ifdef MULTI_MODE_DRIVE
      // BEGINNER MODE:     Power ON + Brake [released] + Throttle [released or pressed]
      #define MULTI_MODE_DRIVE_M1_MAX   1000
      #define MULTI_MODE_DRIVE_M1_RATE  450
      #define MULTI_MODE_M1_I_MOT_MAX   I_MOT_MAX
      #define MULTI_MODE_M1_N_MOT_MAX   350

      // INTERMEDIATE MODE: Power ON + Brake [pressed] + Throttle [released]
      #define MULTI_MODE_DRIVE_M2_MAX   1000
      #define MULTI_MODE_DRIVE_M2_RATE  450
      #define MULTI_MODE_M2_I_MOT_MAX   I_MOT_MAX
      #define MULTI_MODE_M2_N_MOT_MAX   200

      // ADVANCED MODE:    Power ON + Brake [pressed] + Throttle [pressed]
      #define MULTI_MODE_DRIVE_M3_MAX   1200
      #define MULTI_MODE_DRIVE_M3_RATE  600
      #define MULTI_MODE_M3_I_MOT_MAX   I_MOT_MAX
      #define MULTI_MODE_M3_N_MOT_MAX   N_MOT_MAX
  #endif