EFeru/hoverboard-firmware-hack-FOC #416: BMS overdischarge triggering ?

Clarepz, 2023-06-14:
### Variant

ADC

### Control type

FOC

### Control mode

Torque

### Description

I'm making a drift trike with a classical 3 wire ebike throttle as an input. The board shuts down without bipping when accelerating to fast, and cannot be powered again using the power button. I need to unplug and replug the battery to be able to power up the board again. I assume that the over discharge protection of the BMS is being triggered. Am I right ? 

Is  limiting the current more my only option ?

But I didn't change the default 15 A current limit setting. If I'm right, How can it be that the stock hoverboard BMS current limit is lower than 15 amps ?

Candas1, 2023-06-14:
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Battery#if-the-board-turns-off-without-the-melody-the-bms-is-shutting-the-power-off

farev, 2023-09-12:
I am having the same problem with a hoverboard that I haven't reflashed. Is there a way to prevent the overdischarge protection to trigger or should I just use a more powerful battery? Also would connecting two hoverboard batteries together in parallel work?

Clarepz, 2023-09-12:
the thing that worked for me was changing the battery with another one from an other hoverboard

farev, 2023-09-12:
I already tried using a battery from another hoverboard but I got the same problem. I guess it depends on the quality of the manufacturer.