EFeru/hoverboard-firmware-hack-FOC #418: USART control stopped working

farev, 2023-06-19:
### Variant

USART

### Control type

FOC

### Control mode

Voltage

### What is the bug and how to reproduce it ?

I made a go kart a few years ago using this firmware, I control it using an Arduino. Everything was working fine until recently. I just changed the motors/wheels and the battery of my go kart but when I turned it on again it didn't work. I am getting a serial communication timeout from the board (3 beeps). I already made sure none of the cables are the problem and I also tried with another Arduino board but I still get the same error. I didn't change anything besided the motors and the battery, although the go kart did spent several weeks powered off. I am pretty sure all the components are fine so I am not sure why it is no longer working.

Candas1, 2023-06-19:
Hi,

The firmware might have changed in the meantime...
So it could be it's not expecting the same data frame now.
You need to compare the arduino code from the repository with yours.

farev, 2023-06-19:
I compared it and the arduino code is still the same. Also the problem appeared before I even uploaded the new firmware to the motherboard.

Candas1, 2023-06-19:
Ah I think I misunderstood.
But why you flagged it as a bug then ?

farev, 2023-06-19:
I wasn't sure what to label it as.

farev, 2023-06-19:
Yes I checked them but none of them worked.

farev, 2023-06-20:
I already tried doing a full chip erase and flashing the firmware again but I still get 3 beeps and it doesn't work.

Candas1, 2023-06-20:
If [this](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-USART#toolbox-troubleshooting) doesn't help you, I cannot help you.

Either you are doing something wrong, or your hardware has a problem.

TsarkovMaxim, 2023-07-16:
Hi, did you manage to solve this problem? I have a similar story. I flashed the board for uart and no matter what I change in the firmware parameters, I always get a 3 signal at the output, which indicates an error ... Thank you!

farev, 2023-07-16:
I am actually still trying to solve the problem. I am not sure if it is the new version of the firmware but I will try to find the old version I was using and see if it works with the old version.

farev, 2023-09-12:
I am still getting the error, I already replaced the motherboard on my go kart and flashed it with the uart variant but I still get 3 beeps. I tried switching up the tx and rx wires and that didn't help. Also I don't know why but when I connect either of the tx or rx wires coming from the arduino to the tx or rx wire on the sensor cables from the motherboard the motherboard turns on. I would assume this means that something is being transmitted through the wires. Has anyone else tried to use the usart variant recently? Because it used to work with older versions of the firmware.

farev, 2023-09-14:
I was using the wrong pins on the arduino. I was trying to use the tx and rx pins when I should have been using the normal d2 and d3 pins as labeled on the SoftwareSerial in the arduino code.  Also another thing that was causing some problems is that some of my jumper wires were not working properly. It now works perfectly.

closed 2023-09-14 by farev