EFeru/hoverboard-firmware-hack-FOC #423: Mainboard won’t power on with 56v battery

rockydubb, 2023-07-26:
### Variant

HOVERCAR

### Control type

FOC

### Control mode

Voltage

### Description

Hi guys, 

I flashed the firmware to my main & side boards successfully and I got everything working fine like the pedals & break, lights and everything works as expected with the stock Hoverboard battery.

However only after a few minutes I noticed that the board starts to beep indicating the battery is almost depleted so I wanted to use a few spare batteries I have lying around from my Segway ninebot devices like the miniPRO (now called S), S-Plus … these batteries I believe are 57 V when fully charged. And when used with the Segway devices give 30+ kilometers of range. the Scooters from which they came out of no longer work. 

As per the wiki I replaced all the highlighted caps to the Micro farad rating True that @ +100 V. 

However when I Connect the battery to the board and press the power button I briefly hear the melody and immediately the board Powers off. I tried press no worries that's good that's good let's seeing the power button again and nothing happens.

If I disconnect and reconnect the battery then press power … I again hear the melody followed by the board shutting off and no longer responding.

I also have a 9 aH Dewalt battery that says 20V & 60V on it. I use it with my angle grinder. I tested the voltage with my multimeter and it's at 20 V. When I connected it to the board the exact same thing happened where I hear the melody in the board appears to be not powered an unresponsive until I disconnect and reconnect it.

I can only seem to get the board working with just the hoverboard batteries. I've been going through all the posts and I have even tried updating the config.g battery section updating BATT_CELLS @ 30,  along with the BAT_CALIB_REAL_VOLTAGE @ 5700…. Resulting in same issue remaining.

In some posts some users say they have connected their 67 V batteries without upgrading the capacitors and everything works fine no problems. I tried stock 63 V caps and 100 V Phillips caps … call resulting in same outcome.

Can anyone help me because literally I don't know where else to look. Thank you!🙏🙏

Candas1, 2023-07-26:
Hi,

30 as BAT_CELLS ? 
If BAT_CELLS is 10 for a hoverboard battery (42v), 30 is then a 126v battery ?
You should use BAT_CALIB_REAL_VOLTAGE only if you use the battery calibration as described in the wiki.

If the board is turning off with a melody, it's probably because it thinks your battery is too low based on the parameters you are setting.
You can test with BAT_DEAD_ENABLE = 0 to prevent the board turning off for your tests until you figure out what is wrong.

rockydubb, 2023-07-26:
Hi Ca

> Hi,
>
> 30 as BAT_CELLS ? If BAT_CELLS is 10 for a hoverboard battery (42v), 30 is then a 126v battery ? You should use BAT_CALIB_REAL_VOLTAGE only if you use the battery calibration as described in [...]

Hi Candas1. Thanks for such a fast response.

I put 30 as the BAT_CELLS because I thought the description said to put how many cells are in the battery you're using which in my case was 30. Oh ok I will try your recommendation when I get home and let you know what happens.

Again thank you so much for the fast response.

rockydubb, 2023-07-26:
> Hi,
>
> 30 as BAT_CELLS ? If BAT_CELLS is 10 for a hoverboard battery (42v), 30 is then a 126v battery ? You should use BAT_CALIB_REAL_VOLTAGE only if you use the battery calibration as described in [...]

Hi Candas1, 

So i restored all the values back to default BATT_CELLS to 10, the other value back to 3970), and i changed the BAT_DEAD_ENABLE to 0. 

And now when i connect the other batteries .... nothing happens ... not even a melody. I checked the batteries voltage and both are at over 50v+ ...

Candas1, 2023-07-27:
Then you are doing something else wrong.

rockydubb, 2023-07-27:
> Then you are doing something else wrong. 

Hi, 

All the values and configurations in the firmware are unchanged. Except for me selecting the variant Hovercar in the PlatformIO file ... and the one change you said yesterday. 

I unplugged everything from the board. It only has the power button & charger connector attached to the board. 

If i connect the OEM hoverboard battery the board turns on and stays on indefinitely no problem. But when i connect any other battery 42V+ ... nothing happens .. i can post photo if you would like. But how i described above is exactly how the board is sitting on my table.

Candas1, 2023-07-27:
The button has to be pressed while flashing

rockydubb, 2023-07-27:
> The button has to be pressed while flashing 

Hi, 

Yes i know. I have flashed 2-3 boards and sideboards and i know you have to keep the power button held while its flashing or else it gives an error. 

Firmware flashed on to the board just fine. I forgot to mention that i also disabled the sideboard and enabled the debug serial port also which works perfectly with YAT (im using).

rockydubb, 2023-07-30:
Any ideas on what else i can try? The only battery i can get to work with the board is the stock hoverboard battery ... and it runs out in few mins and does not go fast at all.

hbnewbie, 2023-08-10:
if you havent already figured this out, #define BAT_CELLS is for how many cells are in series.  you should verify voltage with voltmeter and google to see what cell arrangement whatever pack you are using has.  that dewalt 20v likely only has 5 cells in series.  there is no point to use this pack since its only half the voltage of most hoverboard packs. and i dont know what would happen if you did since these motherboards are usually set up for 10s (42volt max). but if you were trying to use that you would need to enter 5 for #define BAT_CELLS.  i have used 13s and 14s packs without issue.  (54.6 and 58.8volts). but havent done anything higher.  one hoverboard pack for anything youre riding on is not enough in my opinion

rockydubb, 2023-08-11:
I tried that along with all the other suggestions by Candas1 ... Nothing ... I tried plugging in another 50v+ battery and nothing happens when i press power ... 

Only when i connect the Original hoverboard battery does the board turn on!