EFeru/hoverboard-firmware-hack-FOC #433: MM32SPIN05PF - I'm screwed, aren't I?

jzellis, 2023-09-01:
### Variant

HOVERBOARD

### Control type

None

### Control mode

None

### Description

Hi folks. I'm a total newbie to hardware hacking - I'm a Node dev - but I found a hoverboard in really decent condition cheap at a pawn shop. I wanted to turn it into an electric wagon to help me haul stuff - I live in London (though I'm American) and I don't have a car; I do however have congestive heart failure, and apparently it's not good for me to haul 100 lbs of lumber or pallets around by myself because I might "overdo it" and "keel over and die". So I thought - well, cool, for £50 I can just modify this thing a bit and use a Pi Pico to turn the motors on, connected to a Bluetooth app or controller, and it'll make it easy for me to actually carry heavy stuff.

Obviously, it is not that easy.

I opened her up and discovered that it's got a splitboard with a chip called a MM32SPIN05PF on it. Googling and looking through issues on here tells me there's probably not actually any firmware for it, even if I knew how to use an St-Link, which I don't - I have used DC two-wire motor controllers with a Pi and a Pico, but this stuff is largely a above my pay grade.

Am I wrong? Will this firmware work? I don't know if the firmware has anything to do with the charger or anything, or if it just controls the motors. Would it be dangerous to try it? Alternatively, could I just get some kind of simple 3 phase motor controller? I have a Pico motor controller hat but it only seems to go up to 24v at most, and it's for DC, and I assume this board's motors are AC.

I'm just really lost and I could use some advice. If it's worth getting an St-Link and trying it out with this firmware I'm willing to scale the learning curve, but if there's an easier way to do this, if anyone knows of one please let me know. I feel like a complete moron right now.

Thanks!

[image]

[image]

Candas1, 2023-09-02:
Hi,

I believe this [project](https://github.com/Kambodia91/Hoverboard-firmware-hack-FOC-MASTER-SLAVE) uses the same boards but he swapped the chips with STM32 because the MM32 have little support and would require a rewrite.

Fish-man134, 2023-09-23:
I can't comment on using those boards but my first mod using hoverboard motors was a powered bike trailer which sounds similar to what you want, it uses two e-bike controllers with the throttles wired together , got a few hundred miles on it and it works great . 

So yes you can use cheap e-bike style controllers to power these motors just a quick search will bring up a multitude of options.

xnaron, 2023-12-30:
Are the MM32 pin compatible with the STM32?  IE: just replace the chip with an stm32?

RoboDurden, 2023-12-31:
Replace with a GD32F130C8T6: https://www.aliexpress.com/item/1005006091456025.html
But you would still need to trace the pins on your new layout.

Better buy some other hoverboard that is as old as possible. Then chances are good that you get a Gen1 board for the EFeru firmware or a Gen2.x that i might support. With my 18 different layouts so far 6 layouts are 99% ready and 5 worth testing.
You should get an old hoverboard (with broken battery) for less then 20 UKP in London i guess.

Or buy two of these, they can be driven with a pwm signal: https://www.aliexpress.com/item/4000585317654.html

But again, my Gen2.x uses the Keil ide which does support this MM32SPIN05PF mcu. But i need someone to help me with adding a new target that gives support for  the adc,uart,etc. peripherials.