EFeru/hoverboard-firmware-hack-FOC #441: Motors stutter and go off while driving

DrKlipper, 2023-10-09:
### Variant

USART

### Control type

FOC

### Control mode

Voltage

### What is the bug and how to reproduce it ?

Hi !

I'm just building a "driveable beer case". The work is almost done, and now it's time to get the control ready. I use the SERIAL communication in tank mode. At the moment, the rest of the code is exact the same as the repo. Controlling is done by a Raspberry Pi Zero W2 and Python. I use a joystick in combination with evdev to get my values for the driving / steering. 

So far all works well but after I drove this "thing" I noticed that the run is not really smooth. So I put all on my desk and while playing with the joystick I noticed this stuttering. I recorded a short video: http://www.drklipper.de/VID_20231009_044156.mp4

Sometimes the motors went off even if the serial command tells them to spin forward. And take a look at video position ~0:22. There is this event where the motors do something like a full stop. 

Can anyone give me some advice / test / hint what I can do now. Because right now this creation is not driveable. 

Dominik

Candas1, 2023-10-09:
Hi,

You need to check if the problem is the communication.
Try with the ADC variant and potentiometers for example.

DrKlipper, 2023-10-09:
Hi Candas1, 

thx for the reply. Ok I will check it with the ADC variant. 
But one other question .. How often can I update the values via UART? 
At the moment, I slow it down to update only ever 250ms at max. 
Is that too often?

Candas1, 2023-10-09:
Please check the wiki it's detailed there. Maybe it gives you more insights