EFeru/hoverboard-firmware-hack-FOC #442: Board shuts down completely under load

DrKlipper, 2023-10-09:
### Variant

USART

### Control type

FOC

### Control mode

Torque

### What is the bug and how to reproduce it ?

Hi !
I'm sorry, but I have another topic which bothers me now quite some days ... 
If I try to drive my "vehicle" in some cases it shuts down completely. 
And I think this is not even a normal power down because I'm unable to switch it on again. 
I have to unplug and plug the power cable in order to get the board up and running again. 

It shouldn't be a battery problem either, it was freshly charged. I don't even have to drive long or fast. This problem always seems to occur when I try to get too much power too fast. 

Is this problem possibly known? Does anyone have any tips on how I can isolate this further?

Candas1, 2023-10-09:
Please also check the wiki for this.
Using the search in the repository will also show matches in the wiki.

mdaraVee, 2023-10-12:
Variant
HOVERCAR

Control type
 FOC

Control mode
TORQUE

I can relate to your experience, and I've been facing a similar issue with my hovercar. It's quite frustrating, isn't it? In my case, I've noticed that the vehicle shuts down when I accelerate rapidly at full speed, and this only seems to happen in intermediate and advanced modes, not in beginner mode. However, it works fine when I accelerate gradually.
Let's work together to get to the bottom of this issue and hopefully find a solution. If you discover any fixes or if you need more information, feel free to share it here. We can collaborate on solving this problem and ensuring a smoother ride for our vehicles.
Consulting the Wiki: I understand that I've checked the wiki for this issue and used the repository's search feature, but haven't found a solution yet.I periodically check for updates.

DrKlipper, 2023-10-14:
Hi !
Well I think the main reason for complete switching off is a bad battery. 
[image]
This was my battery and it was a used one ... I bought a power load in order to get such tests done recently. 
As you can see when I discharge with 7A constant current there is a huge voltage drop within the first 2 minutes. And after 5 minutes we lost nearly 10V. I think the internal resistance of that battery is way too high. This shows also the temperature of the battery. It gets extremely warm after this 17min discharge. 
If I am right this hover firmware limits the current from the battery at ~17A. Considering the bad performance of the battery it´s no wonder that the BMS will shut down the complete battery because the voltage drop at high amps will be even higher. 

So I will try another battery which has a better performance (even if it is also a used one...). 
Will see if that works better.

farev, 2023-10-15:
I was facing the exact same issue with a one wheel I made. But I hadn't even reprogrammed the motherboard and still I had the same issue. So that rules out any problem with the firmware. I do believe this problem is cause by the hoverboards cheap batteries. Although I tried changing the battery to another one (it still was a hoverboard battery) and I still got the same problem, maybe I just got unlucky with both batteries or my one wheel draws too much current. I am in the process of getting a better battery, not one from a hoverboard this time, and see if it work with that one.

DrKlipper, 2023-10-15:
I´m in the lucky position to have some hardware for data logging (from the RC model world). And yesterday I did a new test ride on my beer box. This was the result: 
[image]
This was a very slow ride, but you can see that in some cases the voltage dropped to 30V (and the current rises to 17A). But in general die battery performed much better since I don´t had any switch offs. 
My next step will be to limit the current in the firmware. That should result in smaller voltage drops. So the BMS won´t shut down not so quickly ... 

>  I am in the process of getting a better battery
That is also something I consider. I think some 50C/60C Lipo batteries would be fine. They can handle much more current and overall more reliable. https://hobbyking.com/de_de/batteries-chargers/batteries/lipo.htm
2x5s in series would be fine. And since there is a balancer port you can easily add 2 cell monitors while driving.

T41m00r, 2024-04-02:
Hi, Can I use a 6S battery to power the motors? What is the cutoff voltage of the board?