EFeru/hoverboard-firmware-hack-FOC #454: Velcocity Randomly jumps using UART variant

mandrievich, 2023-12-09:
### Variant

USART

### Control type

None

### Control mode

None

### Description

Hello,

I'm using the latest firmware version, everything works fine, but sometimes (it can happen once every 10 minutes or once every 30 minutes) the wheels start spinning abruptly with a noise for a very short period of time, it looks like a bang and then abruptly disappears. This happens when zero signal is applied and during normal operation.

I found old question #120, but I don't think that's the issue as I'm using the latest firmware version.

Can you please help me to figure out the problem?

Candas1, 2023-12-10:
Hi,

In the thread you shared, he ends up using different hardware.
You have not shared any details about your hardware.

I hope you are using hardware serial.

mandrievich, 2023-12-10:
Hi, I'm using USB to TTL converter as hardware connected to Hoverboard.

I tried to use different USB to TTL but the result the same.

Also i tried to use Arduino nano with sample shared (https://github.com/EFeru/hoverboard-firmware-hack-FOC/tree/main/Arduino/hoverserial) to rule out the possibility that the USB to TTL converter is broken.

With Arduino Nano the result the same.

Candas1, 2023-12-10:
Are you sure it's not this?
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Battery#zzz-inactivity-timeout---power-saving

mandrievich, 2023-12-10:
I think this is not the case because i sat the value INACTIVITY_TIMEOUT  to 30 and then to 900, because in my case I don't need INACTIVITY_TIMEOUT  at all.

Candas1, 2023-12-10:
OK then I don't know

geofrancis, 2024-06-18:
I am also getting this issue on the latest firmware, i have tried the wheels  dont rotate smoothly, they abruptly stop with a bang every few seconds, works fine with pwm and ibus.

Candas1, 2024-06-18:
> I am also getting this issue on the latest firmware, i have tried the wheels  dont rotate smoothly, they abruptly stop with a bang every few seconds, works fine with pwm and ibus.

Ibus is also using usart, just a different protocol. Then the problem is with the  device sending the commands.

Candas1, 2024-06-18:
FYI the wiki says you should reduce the baud rate if using software serial on arduino

https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-USART#toolbox-troubleshooting

geofrancis, 2024-06-18:
im using a RP-2040.

Candas1, 2024-06-18:
> im using a RP-2040. il try with slower serial.

Are you using software serial ?

geofrancis, 2024-06-18:
no PIO https://arduino-pico.readthedocs.io/en/latest/piouart.html

geofrancis, 2024-06-19:
I just tried it with a 328 nano board, as soon as its set to under 100rpm it starts banging and stuttering.

geofrancis, 2024-06-20:
the issue with the RP-2040 was the PIO serial definitions are reversed. so when i changed it from softsserial to PIO it swapped the TX and RX.