EFeru/hoverboard-firmware-hack-FOC #455: `TRQ_MODE` overshooting.

Blue186, 2023-12-09:
### Variant

USART

### Control type

FOC

### Control mode

None

### Description

Hello, I have been trying to implement control mode switching through serial messages using SERIAL_START_FRAME. I observed that when in USART+FOC_CTRL+SPD_MODE/VLT_MODE, everything works fine, and the mode switch is successful. However, when attempting USART+FOC_CTRL+SPD_MODE/TRQ_MODE, after switching to TRQ_MODE, there are no errors, but when I slightly change command_in->steer or command_in->speed, the motor speed increases directly to the maximum.When I use the `TRQ_MODE` mode alone, the same issue occurs when combined with serial communication. I have tried reading through the code but haven't identified the issue. I would greatly appreciate your help if you could assist me in resolving this problem.
- 
[image]
[image]

Candas1, 2023-12-10:
Hi,

It's hard to follow you.
If it's happening with torque mode only, why share all the unrelated details about changing the mode?

This is how torque mode works without load. 

Changing the mode is already covered with a different protocol if you read the wiki.

Blue186, 2023-12-11:
Thanks a lot, yes I found it. I found through issues that the torque mode is not linear and not very adjustable, so can I control the input of the torque mode through PID? Calculate the speed difference of the controlled variable, then I can control the speed in the torque mode, so as to better adjust the torque? (Not FOC speed mode) I will try that later.

Candas1, 2023-12-11:
[This](https://github.com/Candas1/Sideboard-Arduino/blob/master/src/Hoverserial.cpp) project is unfinished but you might be able to use the protocol piece.

For the non linearity, I think it's more a scale problem, it's mentioned on those issues.

Blue186, 2023-12-11:
Ok,thanks!
I'll try to adapt it.
In addition, I would like to ask, when I use a motor with a reduction ratio of 1:4.4 and 10 pole pairs, everything is normal when the motor rotates in the forward direction, but when it rotates in the reverse direction, it will stall and cannot start. ,what could be the problem? Of course, I also modified n_polePairs: 10 in the BLDC_controller_data.c file, and also tried to increase r_errInpTgtThres and t_errQual.

Blue186, 2023-12-11:
> Ok,thanks! I'll try to adapt it. In addition, I would like to ask, when I use a motor with a reduction ratio of 1:4.4 and 10 pole pairs, everything is normal when the motor rotates in the forward [...]

This happens from time to time.

Candas1, 2023-12-11:
Are the phases/halls correct ?

lebedyncrs, 2024-08-13:
> Hi,
> 
> It's hard to follow you. If it's happening with torque mode only, why share all the unrelated details about changing the mode?
> 
> This is how torque mode works without load.
> 
> Changing the mode is already covered with a different protocol if you read the wiki.

Can you please share wiki about changing mode? Cannot find it

Blue186, 2024-08-14:
> > 你好，
> > 很难理解你的意思。如果这种情况只发生在扭矩模式下，为什么要分享有关更改模式的所有不相关的细节？
> > 这就是扭矩模式无负载的工作方式。
> > 如果你读过 wiki，就会发现更改模式已经包含在不同的协议中。
> 
> 你能分享关于更改模式的 wiki 吗？找不到

hello,
The torque mode is indeed not linear. If you need to modify the control mode and parameters first, you can do so in the config file, as shown in the following figure
[image]

closed 2024-08-14 by Blue186