EFeru/hoverboard-firmware-hack-FOC #456: VARIANT_PWM issues on several boards

silentbogo, 2023-12-13:
### Variant

PWM

### Control type

FOC

### Control mode

Voltage

### What is the bug and how to reproduce it ?

Hello, and thanks for all the wonderful work you did on this firmware. 
I'm currently having a hard time making an RC AWD rover. The main idea is to have two boards hooked up to a flight controller (in my case it's a Matek F405 WTE). 
I've got 3 similar boards. 
One is an STM32 variant, which worked semi-successfully. Got both wheels spinning using TANK_STEERING. Works in both voltage and torque control mode. Without TANK_STEERING I only get speed control, but no steering at all. 
Second one is the exact same board, only on GD32. Firmware works, but only one wheel is spinning with TANK_STEERING.
Third one is a bit of a weird one. It's a red GD32 board, but looks exactly like the second one in terms of pinout (only SWD header has two swapped pins). The MCU was dead, so I swapped it for an STM32 from a donor. Same behavior as board #2. E.g. only one motor spins erratically. 

Haven't added caps to PWM1 and PWM2, but just in case I've played around with power(there was too much PWM whine when flight controller is powered directly from 15V output on the header).

Also sometimes either board can go through startup sequence and start jerking motors randomly. Sound beeps kinda indicate that it goes into reboot every couple of seconds, or something like that. 

I'm waiting on my USB-UART converters to arrive any day now, so I can look at debug info, but meanwhile I was hoping someone can point me in the right direction to make this thing working properly.

danielkr71, 2024-01-02:
I'm no expert but I reacted visibly with surprise that you seem to be saying you are feeding the flight controller from that 15V output.  I guess if it is rated to handle it, well ok.  I am using RC multichannel receiver modules, supplied with 5V power I steal from either a pin on a voltage regulator or tapped into the +5V line for the hall sensors (if I recall correctly, my receiver modules were labeled 4V-14V).

silentbogo, 2024-01-02:
The one I'm using is rated up to 6S(25V), but that indirectly contributed to my issue. 
I think the main problem was with PWM wiring. I tested the flight controller with multimeter, and servo ground was tied to the power GND. So with 15V at hand to power my flight controller, I made a stupid assumption that I can skip the signal ground, save some time and wires and everything will work on the common ground, but that ended up not being the case.  After I added an extra ground wire for servo lines - most of the glitches went away. 
I'm still having some other issues with 4 out of 5 of my boards, some may be related to wiring, but others - no idea...
One board occasionally spins right motor at startup, but functions as intended later.
Second and third ones worked flawlessly yesterday, but today only spin left motor(I checked PWM signal, motor wiring, hall sensors etc and everything seems fine). Still had no chance to hook up debug console, though. 
Fourth one is a red PCB which looks identical to the other "green" ones. Replaced the dead GD32 with STM32 from another donor, flashed the firmware, but it only spins one motor as well.

lucasalberto01, 2024-02-29:
I'm having exactly the same problem, with TANK STEERING enabled, only one motor turns, my board is the original one with ST32

asken92, 2024-12-13:
if tank steering is enabled each wheel will be controlled individually. you need a Rc transmitter and reciever that enables channel mixing. then you get differential steering or "tank steer"