EFeru/hoverboard-firmware-hack-FOC #457: 3 beep error after powering on

Dominykasssx, 2023-12-15:
### Variant

USART

### Control type

FOC

### Control mode

Voltage

### Description

Hello there! 

I'm trying to run this firmware first time in my life and I have encouded some problems. I want to control wheels using arduino skech.

I already upload a firmware to the hoverboard mainboard,

Here is the settings:

[image]

[image]
[image]

Info from debug port:

[image]

My mainboard continiously beeps. I think it's a 3 beep code and then small pause.

After uploading skech to the arduino, wheels are not moving. Also tested firmware provided here: https://github.com/lucysrausch/hoverboard-firmware-hack/issues/51#issuecomment-418292550

It still beeps, but wheels are spinning themself, so it means it's not a hardware issue. 

I'm using arduino uno which has which has RX pin 0 and TX pin 1. In provided arduino skech there is defined 2,3 RX,TX pins, so I tried using predefined ones pin at first, later changed to 0,1, but it didn't change a thing. 

[image]

Here is the video/sound of the beep: 

https://github.com/EFeru/hoverboard-firmware-hack-FOC/assets/50581706/b18541aa-d3cd-4bda-b3d2-a0e62ec7934d

What could be wrong? Could not see any solutions for this problem online. Thank you for helping :)

Amitkor12, 2023-12-28:
As per the code the 3 beeps indicates serial timeout duration. try changing this parameter, as per your requirement and check.

Dominykasssx, 2023-12-28:
> l timeout duration

I have wired serial TTL adapter to arduino pins to see if something is being sent. Indeed, arduino sends some data, but not sure if hoverboard mainboard receives it, how can I test this?

Candas1, 2023-12-28:
Anyone reading the wiki?

Amitkor12, 2023-12-28:
> Anyone reading the wiki? 

I have gone through the wiki part and also the code of arduino sketch. But not able to find the exact problem 😕

Candas1, 2023-12-28:
[This](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-USART#toolbox-troubleshooting)?

Dominykasssx, 2024-01-15:
Okay, so I have found the problem. I initially selected USART, expecting it to work with USART3 input by default, but it didn't.

Upon reviewing the config.h file, I discovered that USART2 was set as the default (as I understand). 

[image]

I commented USART2 and uncommented USART3 instead, and now it works fine!

Hope it will help someone.

Dominykasssx, 2024-01-15:
Closing issue, because it's a simple configuration problem

closed 2024-01-15 by Dominykasssx

javiersanzsanchez, 2024-09-03:
@Dominykasssx I hace the same issue, I also uncomment the USART3 but I still having those 3 beeps connecting the Arduino to the left side (long wire). Can you provide any summary about you connections and cone?

carloszambri-lgtm, 2026-09-17:
Mi placa emite los tres pitidos pero tengo el gd32 y creo que es por eso estoy cambiando el Bugs en platformio y mañana lo pruebo no tengo ningún voltaje en usart3  a lo mejor debería cambiar mi Arduino nano por otro que pueda utilizar usart2 ya direis