EFeru/hoverboard-firmware-hack-FOC #465: Torque curve

Amitkor12, 2024-01-29:
### Variant

USART

### Control type

FOC

### Control mode

Torque

### Description

I need to get current values of both motor during runtime.
I tried to get the value on serial terminal from the adc buffer dc_l and dc_r.
Than from this I need to map a time vs current graph.
Not only this but I need to have a torque curve on which I have no idea how to achieve it.
Any hint or idea will be appreciated. 
Thank You.

Candas1, 2024-01-29:
As always the answer is already in the [wiki](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-USART#speech_balloon-feedback)

The webtool has a specific option to read that version of the protocol.
[image]

you could also use a tool like stmviewer to trace any variable, I described in another [forum](https://community.simplefoc.com/t/using-stm-studio-for-variable-monitoring/3177/12?u=candas1) how to use it.

Amitkor12, 2024-01-29:
OK I will try this tomorrow.

Amitkor12, 2024-01-31:
What do these values indicate 
- left_dc_current = 1912
- right_dc_current = 1922 ?
since in code we use [A]*100 --> does this means 19.12 A ?

closed 2024-02-02 by Amitkor12