EFeru/hoverboard-firmware-hack-FOC #467: Main Board Dead

Amitkor12, 2024-02-07:
### Variant

USART

### Control type

FOC

### Control mode

Torque

### Description

I have an STM32F103RCT6 board. While testing it in the above mode, it suddenly stopped functioning. When I attempted to power it on again, it emitted a buzzing sound and immediately shut down. Additionally, the MCU detection by STLink failed. I have consulted the wiki for troubleshooting tips, but none seem to be effective. Any idea what might have gone wrong?

TinkerWorX, 2024-02-07:
Have you tried keeping the power button pressed while connecting with STLink? Could it be shutting down because it isn't configured correctly and thinks the battery is dead?

Amitkor12, 2024-02-08:
I tried the above solution , still no detection of mcu by stlink .
Another observation is , there is different sound produce not the buzz, and the mcu gets heats up .

Candas1, 2024-02-08:
If you do a continuity check between vcc and gnd, it's probably shorted now.

Amitkor12, 2024-02-08:
Yes they are shorted.
so what can be done ?