EFeru/hoverboard-firmware-hack-FOC #470: adding TANK mode

TONYMAKE, 2024-03-07:
### Variant

None

### Control type

None

### Control mode

None

### What can we do to make the firmware better?

...adding TANK mode... i saw some commentend line with "TANK" inside.....

### Describe suggestions or alternatives you have considered

I'm making an airsoft tank, just to enjoy... but i don't know what kind of firmware to use.
I'll use an ELRS radio with 8ch PWM reciver
sorry for my english....
Tony

derFrickler, 2024-09-03:
Sorry, just saw this right now, just uncomment the line for TANK Steering:https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/efb438bf514c9f8c80471f79ab87d2f7099746ea/Inc/config.h#L303 and build the PPM Version. 
Works fine on my Hover Bobcat:

Candas1, 2024-09-03:
Nice build.
Be careful those are plastic wheels.

derFrickler, 2024-09-03:
Thanks, that why i used them for this build, it only runs on 4S and still has sufficient power. 
I keep the good old metal wheels for my other builds: https://der-frickler.net/spass/hovertrike
...and 4WD Pocket Quad to come "soon"