EFeru/hoverboard-firmware-hack-FOC #472: Change control mode with USART signal

matshey, 2024-03-15:
### Variant

USART

### Control type

FOC

### Control mode

Torque

### Description

Hi, 
I have build a hoverboard setup so students can learn the basics of controlling a mobile robot. The setup is controlled with a raspberry pi via USART. 
I would like to know if it is possible in the firmware to make the control mode dependent on a signal sent via USART. This way the control mode could be changed without flashing the firmware.

Candas1, 2024-03-15:
Hi,

Have you looked at the wiki already?

matshey, 2024-03-18:
Hi,

I did look at the wiki! But I only found information about changing the control mode in the firmware before flashing, not about changing the control mode after flashing.

closed 2024-03-27 by matshey