EFeru/hoverboard-firmware-hack-FOC #474: Speed mode pushes the wheels

matisse0966, 2024-03-22:
### Variant

PWM

### Control type

FOC

### Control mode

Speed

### Description

Hi, please tell me how to set the speed mode so that the wheels can be pushed when stationary.I have made a remote control PWM trolley, which is more suitable for driving at a constant speed. I have tried voltage mode and torque mode, but they cannot achieve the required stable speed. Speed mode is very good, but sometimes when you need to push it manually, the speed mode will It prevents me from pushing the car. If I need to push the car, I can only turn off the power. I have tried various parameters. I would like to ask you if I can push the car by hand in speed mode by modifying it. I have read the project questions and encyclopedia. I’ve read a lot and can’t find a solution, thank you

EFeru, 2024-03-22:
If the speed request is 0, it will try to maintain 0 speed, so standstill. I am not sure if there is a simpler way of temporarily disabling speed mode than:
- turn off the system or
- have a button to toggle modes speed mode <-> torque mode

matisse0966, 2024-03-23:
> 如果速度请求为0，它将尝试保持0速度，从而静止。我不确定是否有比以下更简单的方法暂时禁用速度模式：
> 
> * 关闭系统或
> * 有一个按钮可以切换模式速度模式 <-> 扭矩模式

Please tell me how to switch modes. Thank you. I have used the cruise button so far. I checked all the information and found no instructions on switching. If you have time, thank you for your reply.

matisse0966, 2024-03-23:
> * have a button to toggle modes speed mode <-> torque mode

Please tell me how to switch modes. Thank you. I have used the cruise button so far. I checked all the information and found no instructions on switching. If you have time, thank you for your reply.

valerivp, 2024-03-26:
may be set non zero minimal speed? 1 as sample

matisse0966, 2024-03-27:
> may be set non zero minimal speed? 1 as sample

Thanks, I'll look into it and see how it should be done, or do you have any ideas on where to start? In fact, if the speed mode can push and rotate freely at very low speeds or at a standstill, it is very suitable for making a variety of cars.